prusa3d/Prusa-Firmware-Buddy #2202: Relase of Firmware 4.4.0 ?

PhilippeGalid, 2022-06-29:
Please tell me when you will Relase Firmware 4.4.0 with Wifi Support.
There is very long time no Firmware Update for Prusa Mini,is Prusa not interrest to make the Mini better.
For me it looks that Prusa Mini is a dead Projekt now by Prusa ?
Thanks for your hard work, and very good Software and Hardware but so long time no New Fireware is not ok.

Thanks

reedy, 2022-06-29:
While I agree over 6 months since a release is a long time.. The repository is very active - https://github.com/prusa3d/Prusa-Firmware-Buddy/compare/v4.3.4...master with 1,366 commits to master since the v4.3.4 commit.

A roadmap and/or timeline for releases would be good, for sure.

CarlosGS, 2022-06-30:
Yes, they have been working very hard to bring a lot of awesome new features. 4.4.0 is definitely worth the wait :+1:

PhilippeGalid, 2022-07-02:
Soory for my strict words, Josef explained the firmware issue and the USB issue in the last live Youtube blog which is being worked on and will come in the next few weeks. Prusa team thanks for your great products and your great support every day.

closed 2022-07-02 by PhilippeGalid