prusa3d/Prusa-Firmware-Buddy #2240: [QUESTION] Can't build firmware according to instructions

diminDDL, 2022-07-14:
I followed the build instructions (used command `python ./utils/build.py --generate-bbf --final`) and everything worked until it gets to using the python scripts. I get errors like the following:
```
Traceback (most recent call last):
  File "/home/dmytro/Documents/projects/Prusa-Firmware-Buddy/utils/lang.py", line 5, in <module>
    import polib
ModuleNotFoundError: No module named 'polib'
```
Even though I have the modules installed with pip. 
```
pip install polib
/usr/lib/python3/dist-packages/secretstorage/dhcrypto.py:15: CryptographyDeprecationWarning: int_from_bytes is deprecated, use int.from_bytes instead
  from cryptography.utils import int_from_bytes
/usr/lib/python3/dist-packages/secretstorage/util.py:19: CryptographyDeprecationWarning: int_from_bytes is deprecated, use int.from_bytes instead
  from cryptography.utils import int_from_bytes
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: polib in /home/dmytro/.local/lib/python3.8/site-packages (1.1.1)
```
I found #813 but the "solution" proposed there made no difference for me.

laloch, 2022-07-20:
Hi @diminDDL, there seems to be something wrong with your Python environment (perhaps multiple Python versions installed). I'd bet that if you run just `python -c "import polib"`, it will fail in the same way.

laloch, 2022-07-20:
You can check which Python binary the build system does actually use by running `grep "lang.py" src/lang/CMakeFiles/translations-genfiles.dir/build.make` in the project root directory.

Prusa-Support, 2022-07-21:
Hello! Sorry for the initial issue and for the late reply - it likely has to do with having multiple versions of Python running. For example you installed on your system the newest one and the project is running onto an old one. The suspect is that the package is indeed installed, but under the "lib" folder for the other Python version. So the workaround would be setting the Python version on the project itself, or alternatively copying-moving the file from the folder of the "version where it is installed" to the "version you were using in the program" would also work. I encountered this problem on an unrelated project and managed to work around it this way. 

I see it worked now for you, I'm not sure on what made it work today but likely something changed along these lines. Of course let me know in case of more feedback or any aspects to further review. 

Alessandro Pantaleo
Prusa Research

Prusa-Support, 2022-07-21:
Sorry @diminDDL I misread, I read that it didn't produce an error so I mistakenly though that the module not found error was solved, sorry for the confusion from my side! Let me know of course if the resolutions would work on your side, or in case there might be need for me to expand on them. 

Alessandro Pantaleo
Prusa Research

diminDDL, 2022-07-23:
@Prusa-Support I am interested in how one could specify the python version to use for the project without moving the libraries around, since I think it isn't that uncommon to have multiple python versions installed on a system.

Prusa-Support, 2022-07-27:
Hello @diminDDL! The best way I found to do the change would be creating a new project and select from there the Virtual Environment - interpreter  
[image] and then copying the files of your project (main.py, etc.) into this newly created project. In the screenshot, for example, it is a Virtual Environment dedicated to that project, that uses Python 10 as interpreter. 

I was checking modifying it within the same project but it would always take using the Virtual Environment of another pre-existing project, I'd recommend more for the project to have its own dedicated Virtual Environment, doable only with the "new project". 

I'm not sure if there might be more to it but with this way should resolve it: after this, either the package would be installed there already and usable, or it should be possible to install it anew, so working around the initial error in both cases. 

Let me know, in case of anything to further review, feel free to ask. 

Alessandro Pantaleo
Prusa Research

diminDDL, 2022-07-29:
Huh, I don't really use an IDE but I'll try to recreate the same in the console.

diminDDL, 2022-08-08:
I finally had the time to try and run it in a venv and nothing changes I still get the import errors despite the pip packages being installed.

diminDDL, 2022-08-08:
Okay I managed to build it after uninstalling python 3.10 however I would propose to make the installer aware from what python instance it was started and carry that over to all the Cmake variables.

closed 2022-08-08 by diminDDL