prusa3d/Prusa-Firmware-Buddy #2296: [BUG] ESP Firmware Flash Failed

chrishilbert, 2022-07-27:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]

**Printer firmware version** - 4.4.0 Beta 2

**Original or Custom firmware** - [Original/]

**Optional upgrades** - [Filament Runout Sensor]

**USB drive or USB/Octoprint**
USB

**Describe the bug**
Getting failed on all my machines with ESP-01 8266 modules when flashing firmware.  Can't seem to get it to flash any of the, always fails at step 2.

[image]

chrishilbert, 2022-07-28:
Managed to get 4/5 machines flashed.  1 is still throwing this error.  Now only 1/5 machines is online wifi.  Tried reloading credentials, redoing the setup several times.

They're all in the same cabinets within 3 feet of each other.  Wifi on my phone is full bars inside the cabinet as well.

DRracer, 2022-07-28:
@chrishilbert please report the exact ESP version + vendor to the google sheet table linked in the release notes. We'll try to find out what is happening on that particular piece of hardware.

chrishilbert, 2022-07-28:
I will add it.  Problem is I have 5 with the same vendor/esp.  2 flashed fine during the initial setup, 3 failed.  1 is working (with IP), 1 is still not able to flash, and 3 are flashed but offline.

chrishilbert, 2022-07-28:
It's updated.  AliExpress one with big note section.

Prusa-Support, 2023-01-12:
Hello. Is this issue still relevant?

Thanks for adding information to the [spreadsheet](https://docs.google.com/spreadsheets/d/1ql2TmS8O5Bkuf9SRhfP67RRD1isdThyN96f9M2KdSg4/edit?usp=sharing).

If you managed to connect all of the various modules from the same manufacturer, you may want to compare their performances to see if all of them are equally compliant. As a test, you can upload a big file via the browser, and check how long the upload takes.
If some of them still don't work, please consider reaching the manufacturer or the vendor for troubleshooting tips, or to have them replaced.

Michele Moramarco
Prusa Research

Levi--G, 2023-02-09:
For me it was solved after changing the mode in the menu to wifi (it was still on ethernet), i don't know if it actually changed anything or lady luck just had something to do with it :D

Prusa-Support, 2023-02-16:
Thanks for your feedback.
That's curious how the Ethernet mode was interfering. If that was the case, and if you can intentionally reproduce the problem, it may be worth posting it as a separate issue for us to review. I seem to understand that the problem discussed in this issue is not concerning the Ethernet connection but a known bug that is supposedly fixed by now.

It may not be a coincidence though if by any chance you recently upgraded the firmware. 🙂
Our developers recently improved this aspect of the firmware with [FW 4.4.1 (final)](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.1).
Please make sure you are running the most recent firmware.

.

I asked if the issue persists but it seems safe to assume that the most recent firmware rendered this issue obsolete.
I will close this issue because stale and theoretically solved, but please feel free to keep sharing relevant feedback.

Michele Moramarco
Prusa Research