prusa3d/Prusa-Firmware-Buddy #2297: [BUG] Error #12608 4.4.0-BETA2 Firmware

McDin, 2022-07-27:
Trying to install the 4.4.0-BETA2 Firmware to Mini+.

Both Files (Bootloader and Firmware) are on USB Stick. After some minutes it went in Error #12608: Firmware in the internal flash corrupted! Please refresh firmware.

Retry Button restarts the Mini+ but same screen comes back.

What should I do?

darraghbr, 2022-07-27:
This happened to me, I ended up going back to 4.3.4 and then downloading a fresh copy of 4.4.0 onto the USB and reflashing and it seems to have worked, unfortunately it removed all of my first layer offsets and other settings but I had them all written down!

McDin, 2022-07-27:
How did you downgrade? I choose old firmware and put it on usb drive but nothing else the error happens :(

darraghbr, 2022-07-27:
I removed the BETA firmware from the USB, added 4.3.4 and then hit the reset button next to the knob on the front of the machine, rather than turning the switch on and off.

littleneutrino, 2022-07-27:
> 

Same here, Even tried different flash drive still get the same. putting old firmware on drive does nothing.

darraghbr, 2022-07-27:
Taken from the [Firmware Flashing Guide](https://help.prusa3d.com/article/firmware-updating-mini-mini_124784):

Alternative method - older firmware release

It is not possible to flash to the same, or an older version of the firmware release, using the method above. To flash an older firmware the procedure is as follows:

    Push the reset-button, immediately followed by double-pressing the selector-knob.
    On the new screen, confirm the update by selecting the "FLASH" button and pressing the knob. 

If it repeatedly only boots the printer and gives you the home-screen try to [format your USB flash-drive](https://help.prusa3d.com/en/article/sd-cards-and-usb-drives_112291) or try a different one. Even if the flash-drive works with G-Codes and printing, it may not work for flashing the firmware.

Maybe try that?

McDin, 2022-07-27:
> I removed the BETA firmware from the USB, added 4.3.4 and then hit the reset button next to the knob on the front of the machine, rather than turning the switch on and off.

Only short press or long? Did you do it when the Error is on screen? For me it reboots in error than again.

McDin, 2022-07-27:
> Taken from the [Firmware Flashing Guide](https://help.prusa3d.com/article/firmware-updating-mini-mini_124784):
>
> Alternative method - older firmware release
>
> It is not possible to flash to the same, [...]

You are are hero, that worked. But I have to hit the knob 3 times. Then at first I got an file system error. With other stick I get asked if I want to flash 4.3.4. Now I have a new try with 4.4.0

littleneutrino, 2022-07-27:
the button pressing worked and im updated now.

McDin, 2022-07-27:
I'm updated, too. But all settings are lost.

McDin, 2022-07-27:
Maybe some knows it. Everything works fine for now. But I can’t see any previews in web Interface and on printer itself. In octoprint I had them. Is this still in progress or a slicer confit? I’m not sure if I changed something in the past for it.

DRracer, 2022-07-28:
@McDin I'm happy that in your case the issue has been "just" a USB bug in the bootloader and that the printer didn't get "bricked". Originally, your report during the Live Stream looked scary.

Previews are normally read from the G-code files if they are there (+have the correct pixel dimension). We didn't have any issues with them during testing, but I'm not saying there are no edge cases. If you can, please share one of your G-code files which is not showing the thumbnail in PrusaLink.

McDin, 2022-07-28:
> @McDin I'm happy that in your case the issue has been "just" a USB bug in the bootloader and that the printer didn't get "bricked". Originally, your report during the Live Stream looked scary. [...]

Trust me I was very scared that it is bricked. But fine it works now.

I put some Screenshort and the gcode in zip here: [https://mcdin1988.synology.me:4001/sharing/l0sF31CYF](url)

DRracer, 2022-07-28:
@McDin good, that's what I thought: you are using different thumbnail sizes (I managed to download the archive before you updated your post :wink: ). MINI only recognizes pixel size 220x124, so the G-code must contain a line like this:
`; thumbnail begin 220x124 ...`

DRracer, 2022-07-28:
closing this issue, thanks :+1:

closed 2022-07-28 by DRracer

McDin, 2022-07-28:
> @McDin good, that's what I thought: you are using different thumbnail sizes (I managed to download the archive before you updated your post 😉 ). MINI only recognizes pixel size 220x124, so the G-code must contain a line like this:
> `; thumbnail begin 220x124 ...`

So 16x16 is unnecessary? Can I delete this from setting?

DRracer, 2022-07-28:
16x16 will be used in the future, keep it there :wink:

sleeve, 2022-09-05:
@DRracer @Prusa-Support I also ran into this issue and noticed that it's still being referenced in a few other tickets (#2409 and #2311). Is it actually fixed in the next beta3 or should it be left open until it's actually fixed? It was an unpleasant user experience seeing the error when upgrading and then having to find the actual double-click workaround fix mentioned above.

Prusa-Support, 2022-09-21:
As mentioned in issue https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2409#issuecomment-1234567973, we may need to collect more data for an accurate reproducibility of the issue, moreover, an investigation is currently open in a different issue.

Similar issues may happen for various reasons and therefore these are hardly trackable and reproducible.
Please notice this issue and the aforementioned one are closed. However, the investigation is still open on the issue https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2311. You can participate in that thread.

By the way, we have a [guide](https://help.prusa3d.com/article/firmware-updating-mini-mini_124784) for this. Please don't hesitate to reach our Customer Support in case of doubts, as it can be of help in this case.
https://help.prusa3d.com/article/customer-support_2287

Michele Moramarco
Prusa Research