prusa3d/Prusa-Firmware-Buddy #2298: [BUG] 4.4.0 - 409 conflict - Can't start print now

darraghbr, 2022-07-27:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]

**Printer firmware version** - 4.4.0 BETA 2

**Original or Custom firmware** - Original

**Optional upgrades** - Filament Runout Sensor

**USB drive or USB/Octoprint**
USB

**Describe the bug**
I log into Prusa Connect using the IP of my printer, connected via Ethernet, log in using the API key.

I go to 'storage' and I select a file that was already on the USB that I have printed before, I click on it, press 'Start Print', Confirm printer ready and I get this pop up in the bottom right of the screen:

[image]

I have also tried generating a new gcode file and uploading it through the UI, I can upload the file but I cannot print the file. 

**How to reproduce**
Attempt to start a print on the USB stick through the web UI. I have tried using both Firefox and Chrome

**Expected behavior**
The print would start

darraghbr, 2022-07-27:
It looks like this is caused by the Prusa being in one of the menus, when I navigate the printer back to 'home' and then try to print through the web UI it works. 

I believe it was the menu where it shows the API key. 

Ideally it would start the print regardless of what menu the printer is in?

art-0, 2022-07-27:
from the release description:

> *MINI+ must be on the home screen, “print-preview” screen, “print-finished” screen - this is intentional for safety reasons

JohnnyDeer, 2022-07-27:
I'm closing this issue as invalid.

closed 2022-07-27 by JohnnyDeer

darraghbr, 2022-07-28:
As per my second comment.

"This can also happen if you start printing the wrong file by accident, hit cancel, go to print the correct file it will give the same error and you will need to manually navigate the printer back to 'home' before starting the print."

Unless "print-cancelled" is a different screen from the "print-finished" screen? (I don't know because this isn't written down anywhere)

After doing some more tests I cannot start a print from the "print-finished" screen either.

tokamac, 2022-09-15:
> from the release description:
> 
> > *MINI+ must be on the home screen, **“print-preview” screen, “print-finished” screen** - this is intentional for safety reasons

Nope, only works from the home screen. For example, when a print has finished properly and all has been removed from the bed, if one has forgotten to click on the main button on the printer to bring back the Home screen, no other print can be started from PrusaLink on the distant computer (which displays the "409: Conflict - Can't start print now" error).

One should be able to launch a new print after a successfully finished one, even if the UI on the printer has not been been manually returned to the Home screen.

Prusa-Support, 2022-09-23:
According to this https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2301#issuecomment-1197709680, actually, the "print-finished" screen may not work as intended and, most importantly, our developers are informed.

For neater communication, please monitor the open issue #2301, while this one remains closed.

Michele Moramarco
Prusa Research