prusa3d/Prusa-Firmware-Buddy #2299: [BUG] Screen flashes while changing filament

mweisshaupt1988, 2022-07-27:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]

**Printer firmware version** - [4.4.0.]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [Filament Runout Sensor, Bondtech Extruder]

**USB drive or USB/Octoprint**
  USB

**Describe the bug**
  After unloading the currently loaded filament the screen starts to flash, the menu 
works but somewhat slower than usual and this problem stops after loading another filament.

**How to reproduce**
  Select "Change Filament" in the Filament menu

[dump.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/9203940/dump.zip)
https://user-images.githubusercontent.com/3426126/181382527-a702174f-0e59-4b11-b022-0c73d12b2970.mp4

JohnnyDeer, 2022-07-27:
@mweisshaupt1988 thank you for reporting. This bug will be fixed in next release.

vanboom, 2022-10-16:
FYI I am experiencing this on my new MINI+ with the 4.4.0-RC1 build.

Prusa-Support, 2022-10-22:
Thank you for reporting. This is not the final release so our developers may still be working on this.
I'm adding the relevant label for more clarity and better traceability of the bug.

Michele Moramarco
Prusa Research

Prusa-Support, 2022-11-26:
The issue should be fixed in the final [FW 4.4.0](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.0) release now. Please check it out.

Michele Moramarco
Prusa Research

closed 2022-11-26 by Prusa-Support