prusa3d/Prusa-Firmware-Buddy #2301: [BUG] 409 - Can't start print now

chrishilbert, 2022-07-28:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]

**Printer firmware version** - 4.4.0 Beta 2

**Original or Custom firmware** - [Original]

**Optional upgrades** - [Filament Runout Sensor]

**USB drive or USB/Octoprint**
USB

**Describe the bug**
Files uploaded from PrusaSlicer often throw 409 can't start print now error in the slicer and the web GUI if kicked off manually.
[image]

**How to reproduce**
  Upload print to printer with "start job" selected.  Get 409 error both in PS and web GUI.

DRracer, 2022-07-28:
@chrishilbert is your MINI really showing the home screen? We only allow starting print while specific screens are being shown on the LCD. Error 409 means your MINI was not "idle" on the home screen.

darraghbr, 2022-07-28:
I'm afraid I can confirm this issue, I cannot start a print from the so called "print-finished" screen. I get the same error, further to my issue #2298 which was closed somewhat prematurely. 

I hope we can all agree that this is a "print-finished" screen?

[image]

I get the below when I try to start a print.

[image]

DRracer, 2022-07-28:
@darraghbr yes, this is a "Print finished" screen. Could you please try starting a print remotely while the MINI is showing the Home screen?

darraghbr, 2022-07-28:
@DRracer That does indeed work as expected.

DRracer, 2022-07-28:
ok, so only the print finished screen seems to be "broken"

darraghbr, 2022-07-28:
Any idea if the "print-cancelled" screen is different mechanically from the "print-finished" screen? If they are different it would be nice to be able to print from the "print-cancelled" screen as I mentioned in my other issue in case you accidentally start to print the wrong file.

chrishilbert, 2022-07-28:
I think I have to hit HOME on the mini before I an start the next job.

chrishilbert, 2022-08-01:
Anyone figure out a workaround for this one?  This is the most annoying bug I have.  I send the print from Prusa Slicer to an idle machine at Home Screen, and have to go to the machine and click "print" on the model up/ready-to-go.

joshnoe94, 2023-10-30:
Any progress on this? Admittedly it's not a showstopper, but it is fairly annoying. Would it be feasible to automatically return to the home screen a few minutes after a print completes?

I don't have any experience with this codebase, so I'm not sure how complicated this would be to fix. If it's something fairly simple, point me in the right direction and I'll take a look.

closed 2024-06-19 by github-actions[bot]

sebastianludwig, 2024-06-19:
@Prusa-Support can you re-open this issue? It might not have had recent discussions but that's more due to the fact that everything has been said, not that it's not an issue anymore. Thanks 👍

danopernis, 2024-06-19:
Before reopening, @sebastianludwig can you please confirm if the issue is still present on newest firmware 6.0.2?