prusa3d/Prusa-Firmware-Buddy #2302: [ENHANCEMENT] Allow password manager to fill API key field on Prusa Connect

darraghbr, 2022-07-28:
It would be nice if the field that shows up when you load the web UI could be filled with a password manager plugin (bitwarden etc)

[image]

Because it is an API key it is probably best kept in an appropriate keystore such as a password manager, and it would be nice if it could be autofilled by said password manager. 

[image]

A very small enhancement but may be nice, especially if you had a number of these machines!

If this is merely a temporary solution and will not work this way once the full version of Prusa Connect is up and running then I can understand if this is not necessary.

chrishilbert, 2022-08-01:
I wish it would just remember that I provided the API key once.

Lichtjaeger, 2022-08-01:
Agree. It would be fine if the page would store the key in the Local Storage or Indexed DB of the browser.

Prusa-Support, 2022-08-31:
Thank you for your comments.

This issue should be listed in the Prusa Link repository but we already have such issue recorded there.
https://github.com/prusa3d/Prusa-Link-Web/issues/222

Please allow me to close this one as a duplicate and feel free to join or monitor the conversation on the issue mentioned above.

closed 2022-08-31 by Prusa-Support