prusa3d/Prusa-Firmware-Buddy #2303: [BUG] ESP Firmware Flash and Wifi Settings Transfer Succeeded But No Connection

skastenholz, 2022-07-28:
**Printer type** - MINI

**Printer firmware version** - 4.4.0 Beta 2, Bootloader 2.0.2

**Original or Custom firmware** - Original

**Optional upgrades** - Filament Runout Sensor

**USB drive or USB/Octoprint** - USB flash drive

**Describe the bug**
 Prusa Mini does not connect to the configured wifi and displays no wifi connection information.

**How to reproduce**
 Follow the steps described in https://help.prusa3d.com/guide/wifi-mini_316781 and failing in step 19 - even after waiting for more than 15 minutes. In addition:

- I exchanged the installed ESP01S from AZDelivery (https://www.amazon.de/gp/product/B074RLG5TC) with another instance in order to exclude individual hardware issues. No success.
- I changed from home router wifi to mobile hot spot in order to exclude wifi connection issues. No success.

**Expected behavior**
Prusa Mini connects to the configured wifi and PrusaLink is reachable via Web-UI.

dstasiuk, 2022-07-28:
The same issue with my purse mini and esp.
Switching router to WPA or WEP does not help.

chrishilbert, 2022-07-28:
Same issue.

PhilippeGalid, 2022-07-29:
Same issue.

padigree, 2022-07-31:
Same for me. Tried with different Devices, Main Network, Mobile Hotspot, Guest Network, Different SSIDs (with or without "Special Characters") But nothing seems to work. Reflashed Both, Printer and ESP - No changes. Can't connect to WiFi.

landmaj, 2022-07-31:
Same issue, the printer will not connect, no matter what. Are there any logs I could check to see what's happening?

pewchai, 2022-07-31:
Had the same issue. Tried two ESP's from the same manufacturer (HiLetgo), nothing. Then got one of the confirmed (DIYmall) on here [WIFI ESP compatibility](https://docs.google.com/spreadsheets/d/1ql2TmS8O5Bkuf9SRhfP67RRD1isdThyN96f9M2KdSg4/edit#gid=0) and it works now. Make sure that the Wi-fi SSID that you're trying to connect to is 2.4 GHz. You might need to go into your router's settings to set it up. 
I recommend reading this for more information on the update: [4.4.0-BETA2 Firmware for Original Prusa MINI/MINI+](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.0-beta2)

Oxize, 2022-08-01:
Here also no luck with ESP-01S from Arceli (Amazon.nl). Somehow it half authed and never connected and stopped working after that.

Ordered an ESP-01S from Levendig. Let you know if that one works.

ZonkSec, 2022-08-05:
I am having this issue as well with a HiLetgo ESP-01 (US Amazon 3pack: B010N1ROQS). I have ordered a DIYmall. Will post my results here.

ZonkSec, 2022-08-06:
My DIYmall ESP-01S (US Amazon: B00O34AGSU) has arrived and I am now able to connect! I don't know why, but the HiLetgo ones did not work for me. Hope this helps others!

landmaj, 2022-08-10:
I bought a module from AI-Thinker and it's working now.

tokamac, 2022-08-11:
Same issue here, with a **Buddy version 2020 rev 1.0.5** (that is the same as in the pictures of the Wi-Fi howto on Prusa's web site!) and the ESP-01S board from _IGMOPNRQ module Store_ on AliExpress that is listed as being "OK TO USE" in the [BOM](https://docs.google.com/spreadsheets/d/1ql2TmS8O5Bkuf9SRhfP67RRD1isdThyN96f9M2KdSg4/edit#gid=0).

I have an eero 5 internet router at home that can temporarily disable its 5GHz band and rely on the 2.4GHz band only, for 15 minutes. I tried this also, to no avail. I note however in the eero FAQ:

> eero only supports WPA2 PSK AES network security, the most secure standard currently available. Additionally, eeros will only send information to our backend services, and our services will only accept information from eeros. This adds an extra layer of security and ensures data isn’t compromised while in transit.
> 
> eero doesn’t support WEP, WPA, or WPS, as these protocols are known to be insecure. This may cause some incompatibility with older hardware that only supports these protocols (such as some [printers](https://support.eero.com/hc/en-us/articles/208114916-How-do-I-connect-my-printer-to-my-eero-network-)), but we firmly believe your network security shouldn’t be compromised.

So I tried adding `WPA2` instead of `WPA` to the line `key_mgmt=` of **prusa_printer_settings.ini**
It does not work either.
At this point I just believe that Prusa didn't implement WPA2 in their code. There is no match doing a research with this term in this sub BTW.

tokamac, 2022-08-11:
> I bought a module from AI-Thinker and it's working now.

Does your internet Wi-Fi router support WPA encryption? Or only WPA2 like mine?

landmaj, 2022-08-11:
[image]

It does but I have it set to WPA2 with AES and it was clearly an issue with the module, not WiFi settings.

tokamac, 2022-08-11:
Thanks, so you do confirm that your newer AI-Thinker module works on your WPA2 network with just `key_mgmt=WPA` written in **prusa_printer_settings.ini**?

landmaj, 2022-08-11:
Yes, ESP-01S with 8Mb Flash from Ai-Thinker works, at least in my printer with my router (TP-Link Archer MR200) with the settings shown above.

Prusa-Support, 2022-09-05:
Hello! Thank you for your reports and sorry for the late response from my side. In case, feel free to fill in new rows with the feedback on possible incompatibility on this community Google Sheet: https://docs.google.com/spreadsheets/d/1ql2TmS8O5Bkuf9SRhfP67RRD1isdThyN96f9M2KdSg4/edit#gid=0 - for example, the Arceli was marked as incompatible for now. 

I am looking for more details on a possible log - a memory dump is possible to generate but at the moment I am unsure it would have info on connection, I'll pull this info as soon as I can. 

In case of anything more to add, feel free to add more comments - I'll update here if I have any more info on this. 

Alessandro Pantaleo
Prusa Research

Oxize, 2022-09-05:
https://docs.google.com/spreadsheets/d/1ql2TmS8O5Bkuf9SRhfP67RRD1isdThyN96f9M2KdSg4/edit#gid=0 - for example, the Arceli was marked as incompatible for now.

I added the ARCELI from Amazon.nl. I even checked it later when i got a new one from Levendig, but the ARCELI wont working.

Oxize, 2022-09-05:
> https://docs.google.com/spreadsheets/d/1ql2TmS8O5Bkuf9SRhfP67RRD1isdThyN96f9M2KdSg4/edit#gid=0 - for example, the Arceli was marked as incompatible for now.
> 
> Alessandro Pantaleo Prusa Research

I added the ARCELI from Amazon.nl. I even checked it later when i got a new one from Levendig, but the ARCELI wont working.

CarlosGS, 2022-09-13:
Hi, I'm also debugging this. Changing parameters in sdkconfig doesn't make any difference.
However, an ESP-01 module that is **not working at the printer**, instead **works well when connected to the computer via tap interface** (using https://github.com/prusa3d/Prusa-ESP-Nic-FW#test-on-linux, same access point & location).
I've also tried dropping baudrate to 1000000 in both sides and it doesn't make a difference either.

Anybody got other clues?

PS: Note the google sheet seems to be deleted, I hope you can recover it as it was very useful.

Prusa-Support, 2022-09-13:
Hello, thank you very much for the feedback, all noted. @CarlosGS I am checking in more detail about the Google Sheet, will let you know here! 

Alessandro Pantaleo
Prusa Research

Prusa-Support, 2022-09-29:
Hello! We fixed the issues on the table, now again regularly accessible. 🙂 @CarlosGS 
Can you please double-check if the model that didn't work on the MINI was already flagged as not compatible in the community table? Or in case, can you please report the findings also there? We'll try understanding it with devs especially if not flagged already as non-compatible. Thank you! 

Alessandro Pantaleo
Prusa Research

CarlosGS, 2022-10-01:
Thank you! Unfortunately I don't remember where I got this ESP from.
I've tested the RC1 and the same issue occurs :(

The module works well when connecting through a computer as a tap interface so it must be due to differences between the Python implementation and the C implementation.
For instance it could be caused by different ESP bootloaders with some startup messages that are properly discarded by the python implementation but mess with the C implementation :thinking:

CarlosGS, 2022-10-08:
So... literally touching & holding the ESP would make it work :facepalm: :laughing:
It seems two flash pins were floating, and setting QIO as the flash mode fixes the problem https://github.com/prusa3d/Prusa-ESP-Nic-FW/pull/10 :)

It was quite a battle but we're headed to support all versions of ESP01! :tada:

CarlosGS, 2022-10-18:
Update: This was a very obscure bug! it turns out to be triggered by slight differences in RF antenna tunning for different ESP-01 modules (see thread in https://github.com/espressif/ESP8266_RTOS_SDK/issues/1200).
Interestingly the problem didn't appear with an older version of Espressif SDK, so it is actually a platform bug on their side.

In the meantime, a patch is to reduce wifi TX power on these modules, which reduces interference and finally allows them to work https://github.com/prusa3d/Prusa-ESP-Nic-FW/pull/11 :tada:

Prusa-Support, 2022-10-19:
Hello @CarlosGS, thank you very much for the update and for sharing the link with the findings, all noted for our devs in case this might help with similar cases. 🙂 

Alessandro Pantaleo
Prusa Research

CarlosGS, 2022-10-28:
Well it is working flawlessly so far :tada: here is the patched firmware in case more people want to test the fix: [esp_wifi_for4.4.0-RC1.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/9886782/esp_wifi_for4.4.0-RC1.zip)

To flash the ESP module, until this is merged, you'll need an "ESP01 USB adapter" and the following command:
`esptool.py -p /dev/ttyUSB0 -b 460800 --after soft_reset write_flash --flash_mode dout --flash_size 1MB --flash_freq 26m 0x0 bootloader.bin 0x8000 partition-table.bin 0x10000 uart_wifi.bin`

Pretty confident on the fix, but being RF voodoo it would be great if more people can verify it. Thanks! :)

DaveMariner, 2022-11-12:
Than's @CarlosGS for discovering this. I can almost confirm this fix, albeit with one degree of separation. 

The TL;DR is that your Espressif thread fixed my similar connectivity problems. Longer explanation follows: 

I have an ESP01-S that refused to connect when I put it in my Prusa Mini+, so I ordered a replacement from a "known good" source, fitted that when it turned up and it's been flawless since. 

The suspect ESP01 was sat on my spares shelf, and yesterday made its way (somehow) into a project I'm working on. I must have picked up the wrong one to program. The chip would connect to the network when sat in the USB programmer, but steadfastly refused when in the project housing. 

The big difference (of course) is the project housing being the circular stainless steel base of a table lamp where I'm using the ESP to drive a LED ring. Space is pretty tight in there, and I'd imagine that  the combination of a 5V PSU and being surrounded by all that metal made it a pretty rich environment for RFI. 

After much tinkering I remembered the issue with my printer. I came here, read the thread, and after reducing the power to 12dBm the ESP connects flawlessly. 

Thanks for pointing me in the right direction!

Eldenroot, 2022-11-12:
This fix did the trick! Works fine with 1.0.5 board

CarlosGS, 2022-11-12:
Thanks! this was very tedious/tricky, so I'm very happy to hear the patch works for you too! 🎉

skastenholz, 2022-11-27:
Thanks @CarlosGS for providing the bugfix. Just want to state that it is not merged yet into the most recent release 4.4.0 and bug decription is updated accordingly.

Oxize, 2022-11-27:
Does it goes official soon?

C44Supra, 2022-12-12:
Did this make it into 4.4.1-RC1? If so, still isn't working for me.

chrishilbert, 2022-12-12:
Still asking for API key on login.

CarlosGS, 2022-12-12:
I don't think it did, but they said i'd be as soon as possible when internal testing is done. Hopefully the next release.

micheld93, 2022-12-18:
Hi, 
I have the official 4.4.0, no beta version. I use a Model from AZDelivery but it is not working. The printer cannot connect to my WiFi. It is WPA2 and 2,4 GHz.

Prusa-Support, 2022-12-24:
Hello @micheld93, sorry for the issue. Was the model also on this sheet? https://docs.google.com/spreadsheets/d/1ql2TmS8O5Bkuf9SRhfP67RRD1isdThyN96f9M2KdSg4/edit#gid=0. 

I had a recent case where in spite of being on the sheet, the link to the actual ESP changed, and replacing the ESP helped in that case. Can you please double-check these aspects as a first thing? 

In case, feel free to move this via email with me info@prusa3d.com mentioning me (Alessandro), we can review in even more detail and see if other tests can be done. 

Alessandro Pantaleo
Prusa Research

micheld93, 2022-12-27:
Hi @Prusa-Support, 
thanks again for helping! The 2 of us already had contact via chat and mail. Yes, it was on the list. I ordered another set of AZ-modules and other ESP modules from another company via amazon. The amazon modules worked. I guess that AZ is changing the modules from charge to charge so it seems to be not very sure that AZ boards are working always. I had 2 others from them which were not compatible as well. 1 from amazon but from Az and 1 other article from their online shop.

So, I can print via wifi now with the new amazon-modules (not the AZ ones) but I think it is better to offer "supported modules" via Prusa online shop or have a collaboration with some companies to offer this in addition to the Prusa shop. I would pay 1-3€ more for such a module instead of ordering 10 modules to test and send them back. This is not sustainable. 

Wishes

PS: I added a comment in the list and added the modules from amazon.de I were able to use.

C44Supra, 2023-02-14:
@Prusa-Support has the PR linked been picked up on? What's going on? I've bought a number of these things from the "verified working" list and have so far ended up with a number of ESPs that most certainly do not work. If at all possible I'd like to avoid having to buy a separate flasher for these things. If you could let us know if there's any work being done on this specific issue and what the expected timeline is, that'd be great.

Prusa-Support, 2023-02-28:
Hello, thank you for the message! Unfortunately, we wouldn't have a timeframe for this, though I confirm that the particular PR is accepted and expected to be part of a future FW release. 

Alessandro Pantaleo
Prusa Research

skastenholz, 2023-04-01:
> Well it is working flawlessly so far 🎉 here is the patched firmware in case more people want to test the fix: [esp_wifi_for4.4.0-RC1.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/9886782/esp_wifi_for4.4.0-RC1.zip)
> 
> To flash the ESP module, until this is merged, you'll need an "ESP01 USB adapter" and the following command: `esptool.py -p /dev/ttyUSB0 -b 460800 --after soft_reset write_flash --flash_mode dout --flash_size 1MB --flash_freq 26m 0x0 bootloader.bin 0x8000 partition-table.bin 0x10000 uart_wifi.bin`
> 
> Pretty confident on the fix, but being RF voodoo it would be great if more people can verify it. Thanks! :)

Since this issue does not seem to be fixed any time soon I gave the above mentioned fix a try and unfortuneately did **not** succeed. Also following https://github.com/prusa3d/Prusa-ESP-Nic-FW#test-on-linux showed the same results: Identical ESP01 device works well on PC but fails on PrusaMini. Therefore, I conclude the only way forward is to purchase a new ESP01 from a proven source.

pilotak, 2023-04-22:
i tried bunch of ESP01 with no success, after purchasing the ESP01S it works perfectly

Prusa-Support, 2023-07-25:
Thanks for your feedback.
If you'd like to, you can contribute to the list of ESP modules in this [spreadsheet](https://docs.google.com/spreadsheets/d/1ql2TmS8O5Bkuf9SRhfP67RRD1isdThyN96f9M2KdSg4/edit?usp=sharing).

Michele Moramarco
Prusa Research