prusa3d/Prusa-Firmware-Buddy #2304: [BUG] Ethernet flashes/cannot be enabled

murk-sy, 2022-07-28:
**Printer type** - [MINI+]

**Printer firmware version** - [4.4.0-BETA2]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [Filament Runout Sensor]

**USB drive or USB/Octoprint**
  Please let us know if you print via USB flash drive or USB/Octoprint

**Describe the bug**
When ethernet is enabled, regardless if it's static or DHCP, the IP data/network icon flashes and the printer is inaccessible via network.
Factory reset does not fix the issue. The issue was present on previous firmware too.
ETH being disabled and opening the IP settings also causes flashing.
Note that this is the second printer with unfixable symptoms described in #1807. It is possible that whatever caused that issue is causing long-term hardware failure. The returned faulty board should still be around.

**LIKELY RELATED ISSUES**: 
#1807 - Recurring issues with multiple printers on old versions, unknown if fixed with this version
#1901 - Same printer on previous FW version

**How to reproduce**
Still unknown, possibly faulty board.

**Expected behaviour**
Report hardware issues with ethernet controller.

**Crash dump file**
[2022-07-28 dump mini4.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/9206066/2022-07-28.dump.mini4.zip)

**Video**
https://www.youtube.com/watch?v=btXw5vM2m9w

I can deliver the affected printer in person next week for testing/board replacement.

murk-sy, 2022-07-28:
Update: After completely powering down the printer for a minute and starting it cold, it has started to function.
However, it does not seem to read static LAN settings (which I'm not sure if it even can in this version).
I've set the IP to be assigned by DHCP for now.

Since this is an issue that so far appeared after the printer was connected to the network for over a month, I will close this issue until it appears again (hopefully never).

murk-sy, 2022-08-11:
Unfortunately it appears my network is cursed. The issue has now appeared again on printer 1.

If possible, please provide steps to diagnose or how to provide more data. At this point I'm willing to hook up wireshark inbetween the printer and the switch just to find out what's going on, if there's anything there to even be seen. @Crawlerin-SA @zoltan-l

Prusa-Support, 2022-10-14:
Hello, thank you for the report, sorry for the late response from my side. Can you please contact me via email info@prusa3d.com mentioning me in the subject, and mentioning the GitHub thread number? Any chances the issue was resolved with the newer 4.4.0RC1? It might be indeed a hardware issue and we can review a few more details on the email thread. Thank you! 

Alessandro Pantaleo
Prusa Research

murk-sy, 2022-10-14:
@Prusa-Support I actually haven't had the issue with RC1, and with 4.4.0-BETA2 it ended up appearing only once on one of the printers. I believe I did not cold boot it after upgrading, which could mean some garbage was stuck in ram? I sort of forgot about this issue since then.

I'll close the issue for now. I'll definitely keep an eye out of any network weirdness.

closed 2022-10-14 by murk-sy