prusa3d/Prusa-Firmware-Buddy #2305: [BUG] Filament change display bug

Fliens, 2022-07-28:
**Printer type** - MINI

**Printer firmware version** - 4.4.0 Beta2

**Original or Custom firmware** - Original

**Optional upgrades** - [e.g. Filament Runout Sensor]

**USB drive or USB/Octoprint**
 USB

**Describe the bug**
The screen flashes white when changing the filament.
The parking position doesn't seem right too.

**How to reproduce**
Change filament

**Video**
(https://user-images.githubusercontent.com/35639879/181437701-42363c7d-6cd3-4690-9048-fc4725488583.mov)

murk-sy, 2022-07-28:
Duplicate of #2299 
New parking position is intentional to reduce strain on PTFE tube, see release notes

JohnnyDeer, 2022-07-28:
I´m closing this issue as duplicate.

closed 2022-07-28 by JohnnyDeer