prusa3d/Prusa-Firmware-Buddy #2310: [BUG] E-Steps etc aren't stored when also editing the footer settings

yodax, 2022-07-28:
**Printer type** - [MINI +]

**Printer firmware version** - [4.4.0-BETA2](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.0-beta2)

**Original or Custom firmware** - [Original]

**Optional upgrades** - [Filament Runout Sensor, Bondtech IFS]

**USB drive or USB/Octoprint**
  USB

**Describe the bug**
  After flashing the new Beta firmware, it failed once. After downgrading and upgrading again it worked but my EEPROM was empty. So I had to redo the e-steps for the bondtech extruder. I saw that the footer settings was in the same menu so I adjusted both. When I did both at the same time the e-steps weren't saved and the printer didn't ask me to reboot.

**How to reproduce**
See above

**Expected behavior**
e-steps etc should be saved even if you modify the footer

Prusa-Support, 2022-08-09:
Hello! Thank you for the report and sorry for the late response from my side, I added the new tag specific for the 4.4.0 Beta and relayed the info to our devs, we'll look into this and fix it as soon as possible. 

Alessandro Pantaleo
Prusa Research

yodax, 2022-08-09:
Hi @Prusa-Support 

I think this [PR ](https://github.com/prusa3d/Prusa-Firmware-Buddy/pull/2373)will take care of it.

yodax, 2022-09-30:
Can confirm working with RC1, closing! Thanks for the fix 👍

closed 2022-09-30 by yodax