prusa3d/Prusa-Firmware-Buddy #2311: [BUG] Beta firmware flash failed, worked after downgrade and upgrade again

yodax, 2022-07-28:
**Printer type** - [MINI]

**Printer firmware version** - [4.4.0-BETA2](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.0-beta2)

**Original or Custom firmware** - [Original]

**Optional upgrades** - [Filament Runout Sensor, Bondtech IFS]

**USB drive or USB/Octoprint**
  USB

**Describe the bug**
  The printer wouldn't start after the initial firmware upgrade. All I got was a message on the screen that the firmware was corrupt. I forgot the exact message. No QR code was given. After downgrading and upgrading again all worked as expected.

**How to reproduce**
  No clue

**Expected behavior**
  Upgrade should work

I don't know how you could do anything with this report, but if there's some extra info I can give you or things you want me to try I'd be happy to help. For now I have a fully working printer, but I thought I'd share anyway.

100jecki, 2022-07-28:
Interesting. I have the same problem. Are you saying that it is enough to re-flash 4.3.4 and then 4.4?

yodax, 2022-07-28:
For me that worked.

edsai, 2022-07-30:
It failed for me as well but I can not get it to flash to the older firmware or reflash the newer firmware from usb. After boot loader 2.02 loads, a red screen that says "Error: 12608 Firmware in the internal flash is corrupted. Please reflash the firmware." pops up with just one option to select which is Reset. Printer is not detected in Prusa Slicer to attempt a reflash from there either.

At the moment, printer is bricked unless there are other options.

yodax, 2022-07-30:
https://help.prusa3d.com/guide/how-to-update-firmware-mini_128421

This tells you how to flash the older firmware

edsai, 2022-07-30:
Thanks for that. It reflashed 4.4.0-BETA2 on it's own because that's what was on the thumb drive and it's working now.

DRracer, 2022-07-30:
Thanks for reporting, we got quite a few reports like this recently. It looks like the bootloader cannot find the appropriate BBF file on the USB flash drive or fails reading it correctly. We'll look into it for the 4.4 final.
The good news is the fact, that the users are still able to flash the old firmware and/or force re-upload of 4.4 beta2 (i.e. the printer doesn't get bricked).

markuswarlich, 2022-08-01:
HELP - HELP - HELP
[image]
[image]
[image]

My total crashed during the upgrade to 4.4.4 alpha. The bootloader is now 2.0.2, but the error message is: "Firmware in the internal flash is corrupt! Please reflash the firmware". I have try to downgrade to 4.3.4 with several USB sticks. It failed. Any help is welcome!!! See attached photos.

yodax, 2022-08-01:
> HELP - HELP - HELP
>
> My total crashed during the upgrade to 4.4.4 alpha. The bootloader is now 2.0.2, but the error message is: "Firmware in the internal flash is corrupt! Please reflash the [...]

Have you tried the downgrade instruction listed here: https://help.prusa3d.com/guide/how-to-update-firmware-mini_128421

markuswarlich, 2022-08-01:
Yes, I have tried it. No success.

markuswarlich, 2022-08-01:
I have use two types of USB-Stick with FAT32 format. One was formatted under MacOS the other under Windows 10. No success at all.

laloch, 2022-08-01:
Hi @markuswarlich, I'm sorry for your problem. Could you please re-download the 4.4.0-Beta2 image from the [Releases page](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.0-beta2), copy it over to a USB flash drive and verify that its MD5 sum is `f73f360b2db68f0171f6ce63a794bd0c`?

markuswarlich, 2022-08-01:
Thanks for your quick response.
Md5 check sum is correct. Format the stick again with FAT32. No luck.
[Ein Bild, das Text enthält.  Automatisch generierte Beschreibung]

Whats next?

Gruß/Best Regards
Markus Warlich

Landgraf-Karl-Str. 12
D-34131 Kassel

mobil: +49 1579 2456202

mail: ***@***.******@***.***>
web: www.warlich.com<http://www.warlich.com>

markuswarlich, 2022-08-02:
Thanks for your advice, but it did not work. I guess the image was written correctly. But the result on the printer was the same as before. Connecting the USB -Stick to the printer and switch it on…. No luck.  The USB stick was not noticed, and the error message is the same as before.

[image]

[image]

After this, I connect the USB Stick again to my mac. It shows:

[image]

Maybe writing the image.dd was not correct?

laloch, 2022-08-02:
@markuswarlich, before we dive deeper into solving bootloader issues, can you please confirm that you do exactly as my GF does on the attached video? Especially the double-tap on the encoder button at the "Bootloader" screen.

https://user-images.githubusercontent.com/1336917/182338089-8bb85c78-38bc-49f5-81b4-3578cb7c2c89.mp4

laloch, 2022-08-02:
The same video on YouTube. GitHub video embedding seems to be broken.

https://youtu.be/8OgkHWrGXTM

markuswarlich, 2022-08-02:
The double-tap was the key to success!! Thanks a lot for all our help here!

laloch, 2022-08-02:
@markuswarlich, great! I'm glad I could help you.

@Prusa-Support, do we have updated instructions for FW recovery/downgrade somewhere? The linked page seems to be outdated.

Prusa-Support, 2022-08-02:
Hi! I noticed we have 2 pages for update and downgrade MINI FW, but the procedure is the same on both pages, but the second and least recently updated page goes in the procedure in more depth: 
https://help.prusa3d.com/article/firmware-updating-mini-mini_124784 
https://help.prusa3d.com/guide/how-to-update-firmware-mini_128421

Alessandro Pantaleo
Prusa Research

laloch, 2022-08-02:
OK, I just verified that the Step 8 from [1] does, indeed, still work. Sorry for the noise.

[1] https://help.prusa3d.com/guide/how-to-update-firmware-mini_128421#128715

markuswarlich, 2022-08-02:
I can confirm after updating 4 Mini's that the error message was shown on all 4 printers. The trick with the double click has worked all the time. Thanks for your support.

JohnnyDeer, 2022-08-15:
Hello, I have a question for everyone who has encountered this issue. 
How many pins does your Buddy Board have next to the Appendix and is there a jumper on those pins? I'm attaching photos of those pins with and without jumper.
Thank you for answers 🙂 
[image]
[image]

yodax, 2022-08-15:
[image]
 Two pins and no jumper

smstroble, 2022-09-03:
Same issue, board version 1.0.5 also just a 2 pin header and no jumper. The double tap during the boot loader screen trick to re-flash the BETA firmware worked without even removing the USB stick.

yodax, 2022-09-30:
Upgrading from BETA to RC went without any issues. Leaving the bug open for @Prusa-Support to close at their discretion.

DRracer, 2022-09-30:
@yodax thank you for reporting your progress, but we still haven't tamed all the USB quirks ... more fixes need to be done into the final 4.4 in this regard.

luastoned, 2022-09-30:
Updated from 4.3.4 to 4.4.0-RC1 and got the same error.

SM-26, 2022-10-18:
Got the same problem,
Updated from 4.3.4 to 4.4.0-RC1 
board version 1.0.5

To solve it, I format the USD drive (just to be on the safe side), and did the double tap on the encoder (this small video was a blessing! thank you!)

Prusa-Support, 2022-10-21:
Hello, thank you very much for the feedback on this, good to hear that this workaround fixed it @SM-26. @luastoned, have you attempted the workaround, did it work for you too? 

Alessandro Pantaleo
Prusa Research

markuswarlich, 2022-10-21:
The latest firmware update worked for all my printers without any issue. Thanks.

luastoned, 2022-10-21:
> Hello, thank you very much for the feedback on this, good to hear that this workaround fixed it @SM-26. @luastoned, have you attempted the workaround, did it work for you too?

Yeah hitting the encoder knob worked after a couple of tries.

stixpjr, 2022-11-28:
fwiw, 3 pins, no jumper, Buddy board 1.0.2. Double tap to reflash from the same image and USB stick worked fine.

areyouready, 2023-01-23:
I downloaded the 4.4.1 final for my mini+ today and got the #12608 as shown above

Prusa-Support, 2023-01-23:
Hello @areyouready, sorry that you had the issue even when flashing 4.4.1. Can you please make these attempts: 

- place the USB flash drive back into the computer. Delete both files and copy them again from the computer to the USB flash drive. Insert the drive back into the printer and restart it. This should initiate the flashing procedure - if not, then you can "force" the flashing using this "alternative method" https://help.prusa3d.com/article/firmware-updating-mini-mini_124784#alternative-method-older-firmware-release. 

- if it doesn't work with this attempt, the issue might be on your USB drive. Can you please attempt it with another USB drive, formatting it in FAT32? https://help.prusa3d.com/article/sd-cards-and-usb-drives_112291#how-do-i-format-the-usb-drive-sd-card

I await your update, let me know how it goes! 

Alessandro Pantaleo
Prusa Research

areyouready, 2023-01-23:
Double pressing the knob and reflashing did. the trick. Thanks for the quick response

Prusa-Support, 2024-01-30:
With [FW 4.4.1](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.1), the flashing operations were made quite straightforward and significantly faster thanks to the bootstrap improvement. As a result, flashing issues decreased drastically immediately after this firmware release.

Now, with FW 5.1.0, it seems safe to officially close this issue as obsolete.

Michele Moramarco
Prusa Research

closed 2024-01-30 by Prusa-Support