prusa3d/Prusa-Firmware-Buddy #2312: [BUG] Wi-Fi Firmware flashing failed

100jecki, 2022-07-28:
[MINI] [4.4.0-BETA2] [2020 rev 1.0.5]
Hey, I just flashed my printer's firmware to 4.4.0-BETA2. Everything was fine.
I wanted to check the WI-FI module. I followed the instructions on https://help.prusa3d.com/guide/how-to-setup-wi-fi-mini-mini_316781
The problem occurred while trying to update Wi-Fi ESP module. On step 2 of 3 i got en ennror "Firmware flashing failed". On another attempt, the screen turned completely white.

How do I get some detailed logs?

100jecki, 2022-07-28:
After doing a downgrade and then flashed the beta version again the screen froze and then glows white. Strange things are happening with a screen.

DRracer, 2022-07-30:
@100jecki screen turning white is really caused by electrical issues (corrupted communication with the LCD). First, please check all the connectors on the board.
Please report the exact ESP module version & vendor into the google sheet table, we'll try to obtain it and see what happened.

100jecki, 2022-08-04:
Successfully flashed after downgrade and upgrade again.

Prusa-Support, 2022-08-04:
Hello @100jecki, thank you for the update and glad to hear that downgrade + upgrade did the trick. Still, feel free to report the ESP module version and vendor on the Google Sheet with the feedback in case. 

Alessandro Pantaleo
Prusa Research

vaskozl, 2022-09-17:
In my case it failed on 0/3 and it was due to inserting the module one row towards the centre - whereby it inserted smoothly but two pins were not in (it really touches the wall of the enclosure when seated correctly). After replacing the module in the correct position it updated correctly.

okaerin, 2022-12-09:
In my case it failed on 2/3. So far I checked that everything is properly inserted and I tried the downgrade/upgrade trick. So far no luck. My mini board revision is 2019 rev 1.0.2 and the datasheet to my purchased module: https://www.taydaelectronics.com/datasheets/files/ESP-01S.pdf

Prusa-Support, 2023-01-12:
Hello. Is this issue still relevant?

Here I seem there was a mixture of factors among electronic/assembly complications, non-compliant hardware, or even bad luck with the installation (ESP flashing should be more reliable with the latest release).

Can the issue be closed?

Michele Moramarco
Prusa Research

okaerin, 2023-01-15:
> Hello. Is this issue still relevant?
>
> Here I seem there was a mixture of factors among electronic/assembly complications, non-compliant hardware, or even bad luck with the installation (ESP flashing [...]

From my side it is resolved, but I don't know about the original issue creator.

puetzcraig, 2023-01-21:
> In my case it failed on 0/3 and it was due to inserting the module one row towards the centre - whereby it inserted smoothly but two pins were not in (it really touches the wall of the enclosure when seated correctly). After replacing the module in the correct position it updated correctly.

You saved me probably hours of troubleshooting - I had made the same error!

closed 2024-06-19 by github-actions[bot]