prusa3d/Prusa-Firmware-Buddy #2314: [BUG] Fan stopped error on low fan speed settings

murk-sy, 2022-07-28:
**Printer type** - [MINI+]

**Printer firmware version** - [4.4.0-BETA2]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [Filament Runout Sensor]

**Describe the bug**
When print fan is set to a low value, it errors despite spinning and being detected as spinning. I use low fan speeds for excellent layer adhesion for PETG.

**Additionally, I have managed to produce a condition where the print fan has an error, but the hotend fan stopped spinning while hotend was at 60+°C**. This was during a print on one machine (fan 36/255) and during manual control trying to reproduce the error on another. Unfortunately I have found no way to reliably reproduce it yet.

**How to reproduce**
Set print fan speed to 40/255 or lower like 8/255, errors should be produced continuously despite the fan being detected as spinning.

**Expected behavior**
On lower speeds, activate error only if the fan is completely stopped or display non-interrupting warnings only.

**G-code**
N/A

**Crash dump file**
N/A

**Video**
Will post if required

jonatino, 2022-07-28:
This happens for me as well and I have not modified fan speeds at all. My prusa is default settings out of the box.

tedmahsun, 2022-07-28:
Stumbled on this as well when trying to print with PETG. Spent hours thinking my fan was faulty, then realised it could be a firmware fault.

tedmahsun, 2022-07-29:
Workaround for now is to disable Fan Check in settings.

chrishilbert, 2022-07-31:
Got this as well!

Prusa-Support, 2022-07-31:
Hello! Thank you very much for the report. Did this occur only with the 4.4.0 beta, or did it ever occur previously to anyone? Disabling fan check can be quite risky, though - if the hotend fan fails, we'd risk having insufficient cooling on the parts, could lead to more serious issues. We'll review the bug and fix it as soon as possible. 

Alessandro Pantaleo
Prusa Research

murk-sy, 2022-07-31:
@Prusa-Support This did not happen before, though I believe the fan is now able to run at lower speeds than before. When I first set up the profile for PETG, I used the minimum fan speed that still worked, which I believe was 36/255. The fan now seems to start even at lower set speeds, which is great but probably related to other code changes.

As I mentioned in the first post, I have had a case where the hotend fan turned off due to this warning - one was during a print (hotend at 245°C!!) and once during reproduction attempts (hotend 60°C). I have not managed to reproduce this yet.

chrishilbert, 2022-07-31:
Can confirm didn’t happen before.  Also only has happened when using PETG

laloch, 2022-08-01:
Hi all, thanks for reporting. @Prusa-Support, I can confirm the issue. We are aware of it and it's already in our backlog.

Spacekoaster, 2022-08-25:
Hello, I've got the exact same behaviour when printing with PETG after upgrading to 4.4.0-BETA2. I can confirm it didn't happen with the latest stable firmware release.

pauhull, 2022-10-02:
This still happens on my printer with [4.4.0-RC1](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.0-RC1) even though on the release page it says that it was fixed.

murk-sy, 2022-10-02:
I think the threshold was just lowered, not set to 0. If you use a fairly low percentage, it will still trigger.

Ideally it should probably be set relative to the set fan rpm, because you can actually run the fan at a very low speed, I believe it starts spinning at about 4/255? It was a while since I tested this.

Probably at a setting below 15% or so it should just check if the fan is spinning at all  (i.e. rpm over 40?) or not, rather than consider the threshold.
The question is more is this intended to check if fan is fully stopped or if fan is spinning too slowly in general.

Prusa-Support, 2022-10-14:
Hello, thank you for the report. @pauhull what sort of speeds did you have on the RC1 when this occurred? I'll note this to our devs. @murk-sy sadly I wouldn't know in more detail, but will review more info with devs to add here. 

Alessandro Pantaleo
Prusa Research

Prusa-Support, 2022-11-26:
The issue should be fixed by the final [FW 4.4.0](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.0) release now. Please check it out.

Michele Moramarco
Prusa Research

closed 2022-11-26 by Prusa-Support

murk-sy, 2022-11-30:
Appears to be resolved - even with 6/255 speed and slowing it down by touching it, it appears to require a significant slowdown before an error is thrown - unfortunately I can't measure it exactly. I've enabled fan check on all printers, will respond again in case of issues.

Prusa-Support, 2022-12-01:
Hi, thank you for confirming that it seems to be resolved, good to hear this aspect is going regularly! 

Alessandro Pantaleo
Prusa Research

thomasvnl, 2023-02-07:
Having this exact same issue with firmware 4.4.1 recently installed and only when started printing with PETG where the print fan is stopped until layer 5 and then as soon as it starts spinning slowly, the printer throws the print fan error.

Prusa-Support, 2023-02-08:
Hello, thank you for reporting this. How much would you need to raise the print fan speed to not have the error? Was it set at 15% from PrusaSlicer? Disabling the fans check is indeed a bit risky, as that would also disable the check for the hotend fan. 

Also, would it occur if you tested the same print with 4.4.0? 

Alessandro Pantaleo
Prusa Research

murk-sy, 2023-09-27:
@larsinka what settings are you using? Did the issue happen with previous stable?

terba, 2024-01-24:
The same problem here with 4.1.1 and 5.1.2 printing ASA on the Mini with the default Prusament ASA profile: first 4 layers off then 20% from the 5th layer. 2 of 3 prints stops at the 5th layer with Print fan error. If I press Continue on the error screen the print goes on without any issue.

Prusa-Support, 2024-02-21:
I wouldn't expect the print fan speed at 20% to cause issues so I doubt it is a problem of firmware or settings.
It may be due to initial mechanical friction that the print fan can't fight at low power, a potential sign of hardware degradation, or due to debris/dust/hair buildups.

Michele Moramarco
Prusa Research