prusa3d/Prusa-Firmware-Buddy #2315: [BUG] BSOD while printing

chrishilbert, 2022-07-28:
**Printer type** - [MINI]

**Printer firmware version** - 4.4.0 Beta 2

**Original or Custom firmware** - [Original]

**Optional upgrades** - [Filament Runout Sensor]

**USB drive or USB/Octoprint**
USB

**Describe the bug**
Got BSOD while printing

**How to reproduce**
Opened web GUI while printing

[image]

chrishilbert, 2022-07-28:
Different machine, got it while hardwired as I just kicked off the job from PrusaSlicer:

[image]

SpeakingOfBrad, 2022-07-28:
I'm having the same issue.

I was able to successfully complete one print sent from PrusaSlicer, but when I tried to start another print I got BSOD (I think it was either during the file transfer, or right after when it started the print).
Then I was in a BSOD loop. After I turn off the printer and turn it back on, it would be okay for ~30-60 seconds before another BSOD. Even turning off wifi from the Networking menu on the printer did not stop the BSOD loop.

It wasn't until I saw this thread and I decided to close the PrusaLink tab I had open on my desktop that the BSOD loop stopped.

Just a note: I discovered while troubleshooting this that disabling the wifi interface from the Networking menu does not make the printer disconnect from wifi (confirmed by the BSOD loop, as well as checking my network router logs). It seems the printer does not disconnect from wifi until you re-enable the wifi interface and then you'll have to reload your network credentials from the usb drive.

[image]

chrishilbert, 2022-07-28:
You beat me to the comment on closing the tab.  Seems to be the WebGUI doing something.

vorner, 2022-08-01:
Hello

We are trying to find out what's different in your case. Can you tell us what OS and browser you're using? Also, does it happen if you access the web from some other device, let's say your phone?

chrishilbert, 2022-08-01:
I was using Safari (latest version) on Mac.  Nothing mobile open.  I've not tried to access it on mobile.

JohnnyDeer, 2022-08-01:
Hi @chrishilbert and @SpeakingOfBrad, can you please attach Crash Dump? You can create dump after BSOD in Settings menu by "Save Crash Dump".
It will help us to debug this issue. Thank you

SpeakingOfBrad, 2022-08-01:
> Hello
> 
> We are trying to find out what's different in your case. Can you tell us what OS and browser you're using? Also, does it happen if you access the web from some other device, let's say your phone?

I am using Firefox 102.0.1 on Ubuntu 22.04.1. I haven't tried to access it on anything else yet. It doesn't happen on every print. Once I was able to exit the BSOD loop by closing the tab, I started another print and it printed just fine. And the printer has been on and idle all weekend with PrusaLink open on my laptop and there hasn't been any BSOD.
I will play with it later today to see if I can get it to BSOD again, and then I'll save the crash dump.

laloch, 2022-08-02:
@SpeakingOfBrad, the printer only remembers one crash dump and it gets only cleared once you save it to the flash drive. So, regardless of how many BSODs you may get in the future, the dump file is going to be the same until you save it.

SpeakingOfBrad, 2022-08-02:
Thanks, I hadn't realized that.
Here is my crash dump.
[dump.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/9242193/dump.zip)

closed 2024-06-19 by github-actions[bot]