prusa3d/Prusa-Firmware-Buddy #2316: [FEATURE REQUEST] Browser notification when print is finished

Fliens, 2022-07-28:
**Printer type** - [MINI]
**Optional upgrades** - [e.g. Filament Runout Sensor]

**Describe the expected functionality**
  When the user has completely set up prusa link, it would be nice if we could get browser notifications for print events.
  With notifications I mean these: https://www.w3.org/TR/push-api/
  And with events I mean events like: print is finished, ready for color change, file upload is finished
  Or any event where the user needs to do something with the printer

chrishilbert, 2022-07-28:
Agreed.  Would be cool to hook on to something like OctoPod as well to push notification of print completion.

Prusa-Support, 2022-09-01:
Hello! Thank you for your request, sorry for the late response. All noted from our side, I passed this on to our devs for them to further evaluate this. In case of more info from our side, we'll update on this thread. 

Alessandro Pantaleo
Prusa Research

Prusa-Support, 2023-10-28:
The new firmware for MINI (newly available in alpha "potentially unstable" version) enables the MINI on Prusa Connect, hence allowing notifications via Telegram, Discord, and Slack - as well as desktop notification via browser.
https://help.prusa3d.com/search?s=connect+notifications+setup

Please allow me to close this issue as solved but feel free to share feedback below.

Michele Moramarco
Prusa Research

closed 2023-10-28 by Prusa-Support

aWZHY0yQH81uOYvH, 2024-07-08:
Would it be possible for this to be built into Prusa*Link* as well? Either have the printer itself POST to a webhook URL or have the PrusaLink page served by the printer use the browser notification API mentioned by OP (seems cleaner).

To me, this would make PrusaLink much more useful.

Prusa-Support, 2024-10-12:
There are several ways to enable notifications but Prusa Link should remain the most basic interface between printer and Prusa Connect. Basic enough to be hosted "on the printer itself".
You can post the suggestion to the repository that is specific to Prusa Link but again, Prusa Link should remain a basic interface with limited options.

Michele Moramarco
Prusa Research