prusa3d/Prusa-Firmware-Buddy #2317: [ENHANCEMENT] As a user I want to see the latest uploaded file to Prusa Link at the top of the file list

tedmahsun, 2022-07-28:
Currently when uploading gcodes to Prusa Link, the file is added to the bottom of the file list. It would be more user friendly for me if after uploading, the file appears at the top instead of the bottom. The file list should also be arranged from latest to oldest, instead of the current way of oldest to latest.

rbronco21, 2022-07-29:
Some sorting method, at least. I prefer alphabetical, and maybe highlight or color the most recent?

Prusa-Support, 2022-08-03:
Hello! Thank you for your request. I noticed other requests with different sorting methods, I forwarded the request to devs if possible to make the sorting method selectable. No guarantees we'd implement for now but I passed the request on. 

Alessandro Pantaleo
Prusa Research

GithubUser99999999, 2022-12-13:
+1

It is really annoying that PrusaLink sorts files alphabetically and the user must resort to file naming tricks to see the latest file on top. If you got many gcode files on that USB drive, it also takes forever to load the previews and they must be reloaded every single time since they're not stored. The USB drive would have more than enough space for that. 

It's also not possible to simply delete gcode files to keep order.

closed 2024-06-19 by github-actions[bot]