prusa3d/Prusa-Firmware-Buddy #2318: Prusa Link cannot display Folders

BenHartwich2005, 2022-07-28:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Mini+** - [MINI]

**4.4.0_Beta2** - [e.g. 4.0.5, ...]

**Original Firmware ** - [Original/Custom]

**Filament Runout Sensor** - [e.g. Filament Runout Sensor]

**Usb Drive**
  Please let us know if you print via USB flash drive or USB/Octoprint

**When using Prusa Link and having folders on the usb stick they dont show up**
  A clear and concise description of what the bug is.

**Just Add an Folder with Files they wont be visible in Prusa Link**
  Please describe steps to reproduce the behavior.

**It Should show the folder**
  A clear and concise description of what you expected to happen.

Emanuelgtxf, 2022-07-29:
[image]

Emanuelgtxf, 2022-07-29:
a min esta assim e da no prusaslicer da erro http 408

Lichtjaeger, 2022-07-31:
I have the same issue.
I have the Revo upgrade and created directories for the different nozzle sizes on the stick. It works fine on the printer. But the web page only shows files in the root and I can't see or access the nozzle directories.

Emanuelgtxf, 2022-07-31:
EU NÃO CONSIGO ENVIAR PELO PRUSASLICER

Emanuelgtxf, 2022-08-01:
alguém consegue saber como o prusa link fica como com o cabo de rede .
eu com  o cabo de rede fica direito .

laloch, 2022-08-02:
Hi @BenHartwich2005, @Lichtjaeger, thanks for the report. You're completely right. Prusa Link is currently designed to search the USB drive for g-code files recursively through all directories on the disk and to display a flat list of all the files found. Unfortunately, the server part of the implementation does not do the recursion as of the current version. So, only files in the root directory are visible for now. Sorry for the inconvenience.

laloch, 2022-08-02:
@Emanuelgtxf, could you please open a separate issue for your problem? It doesn't seem to be related to Prusa Link and folders.

Also, you would probably get better and more timely feedback if you spoke English rather than Portuguese on an English-speaking issue tracker.

===
@Emanuelgtxf, poderia, por favor, abrir um número separado para o seu problema? Não parece estar relacionado com o Prusa Link e pastas.

Além disso, provavelmente obteria um feedback melhor e mais atempado se falasse inglês em vez de português num localizador de questões de língua inglesa.

Emanuelgtxf, 2022-08-02:
but this is also my problem as it is in the image above.
but I can write in with good will to translate as I do when I don't understand.
but thanks the same.

Prusa-Support, 2023-11-30:
Prusa Link, Prusa Connect and the printer's Firmware received several upgrades over the last few months.
The feature should be available and work correctly with the most recent FW 5.1.0, as per https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3117.

Michele Moramarco
Prusa Research

closed 2023-11-30 by Prusa-Support