prusa3d/Prusa-Firmware-Buddy #2320: Display flashing/weirdly refeshing in the filament loading menu

BenHartwich2005, 2022-07-29:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MINI+

**Printer firmware version** - 4.4.0-BETA2

**Original or Custom firmware** - Original

**Optional upgrades** - Filament Runout Sensor

**USB drive or USB/Octoprint**
  USB Drive

**Describe the bug**
  When changing the Filament the LCD starts flashing weirdly when it res

**How to reproduce**
  Just change the Filament mine was PETG 

**Expected behavior**
  Should work normally without the weird refreshing

https://user-images.githubusercontent.com/77836725/181768243-05ed1e11-fa46-4f4c-be82-1afe8d4690d9.mov

tafelberg, 2022-07-29:
I have the same problem.

rbronco21, 2022-07-29:
> I have the same problem.

👍 
I wanted a reaction thumbs up, but I can't get it to work, so you get a reply.

prochazkaml, 2022-07-30:
Same problem. It starts blinking when it asks me whether the filament unload was successful, keeps blinking throughout the entire loading process (including filament selection), it finally stops blinking when it displays "Loading to nozzle".

One strange thing (not sure whether it is a feature or a bug) was that when (un)loading the filament, it rammed and then immediately started pulling the filament out. In the older releases, there was a short pause between the two.

Also, when I select "Retry" after the filament had been loaded, the screen does not flash at all throughout the process and it works fine. The proper pause is now also present between ramming and unloading stages, as was with earlier releases.

Emanuelgtxf, 2022-07-31:
EU TIVE O MESMO PROBELEMA ...

JohnnyDeer, 2022-08-01:
This issue is duplicate of #2299. I´m closing this one.

closed 2022-08-01 by JohnnyDeer