prusa3d/Prusa-Firmware-Buddy #2322: OctoPrint-BedLevelVisualizer - Unfortunately it no longer works with 4.4.0-BETA2

tafelberg, 2022-07-29:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - Mini+
**Printer firmware version** - 4.4.0-BETA2 for Mini

**Original or Custom firmware** - Original

**Optional upgrades** - Filament Runout Sensor, Octopi

**USB drive or USB/Octoprint**
  USB/Octoprint

**Describe the bug**
 
(https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/issues/560#issuecomment-1198951132)

**How to reproduce**
 (https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/issues/560#issuecomment-1198951132)

**Expected behavior**
  A clear and concise description of what you expected to happen.

**G-code**
  Please attach the G-code you had trouble with. This will make it easier for us to replicate the error.

**Crash dump file**
  Please attach the crash dump file. This will make it easier for us to investigate the bug.

**Video**
  Please attach a video. It usually helps to solve the problem.

tafelberg, 2022-08-10:
Hello Allessandro,
same problem at 79 degrees when the target is 80 or 69 degrees when the target is 70.
No Problem or beep without Octoprint. But I can see, that the last degree always needs more time than the other steps before. And often it is switching between the last step and the target for a while. I'm not sure if this was normal and always like this.

Best regards

Prusa-Support, 2022-08-16:
Hello! Thank you for the details, I brought this to our devs attention, we'll review in more detail and find more insights and a solution for this as soon as possible! 

Alessandro Pantaleo
Prusa Research

tafelberg, 2022-10-02:
Hello Alessandro,

still the same problem with 4.4.0-RC1 !!

Best regards

closed 2024-06-19 by github-actions[bot]