prusa3d/Prusa-Firmware-Buddy #2324: Esp8266 firmware flashing failed

gugo2808, 2022-07-30:
[image]
Hello, I have this problem, is it possible to flash esp8266 manually? 
Thank you!

nikomi021, 2022-07-31:
I have the same issue. Just reset to factory settings, flash firmware again.

gugo2808, 2022-07-31:
Actually, I have solved the issue. Luckily I have a lot of esp8266. 
I swapped 3 of them and on the one on which was written ai cloud inside everything worked without resetting the printer.

Emanuelgtxf, 2022-07-31:
EU ALTEREI O NOME DA FIRMWER PARA BETA 3 E JA DEU

Prusa-Support, 2022-08-03:
Hello! Thank you for the report, sorry for the late reply but good to hear it was solved in the meantime. @gugo2808 if possible, can you please send us the links to the pages for specs of the ESP8266 you tried? And in the ones that didn't work, if possible to test, would the solutions proposed by @nikomi021 and @Emanuelgtxf work? Thank you! 

Alessandro Pantaleo
Prusa Research

Emanuelgtxf, 2022-08-03:
mas ainda não funciona so trocai beta 2 para beta 3  mas ainda nao consigo enviar pelo prusa link-.

Prusa-Support, 2022-08-04:
Hello @Emanuelgtxf, thank you for the response. Can you please confirm the exact version of your ESP board, and add the feedback also to this community sheet? https://docs.google.com/spreadsheets/d/1ql2TmS8O5Bkuf9SRhfP67RRD1isdThyN96f9M2KdSg4/edit#gid=0

We'll review in more detail and find a possible solution. Also, was it all OK on the steps here: https://help.prusa3d.com/guide/wifi-mini_316781, including obtaining the IP address and API key from step 20? Obviously without sharing the values. 

Thank you for any feedback! We'll try our best to understand the cause of the issue and resolve it asap. 

Alessandro Pantaleo
Prusa Research

holgerschillack, 2022-08-09:
> I have the same issue. Just reset to factory settings, flash firmware again.

Can confirm that this works for a generic ESP01 with a ESP-8285 chip.

Prusa-Support, 2022-08-09:
Hello @holgerschillack, thank you for this info. In case, feel free to compile it on this community sheet: https://docs.google.com/spreadsheets/d/1ql2TmS8O5Bkuf9SRhfP67RRD1isdThyN96f9M2KdSg4/edit#gid=0, so that we have all the details of the particular ESP that you used. Thank you! 

Alessandro Pantaleo
Prusa Research

Prusa-Support, 2023-01-12:
Hi everyone.
This issue seems to be a duplicate of the issue https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2312.

I hope you don't mind me closing this one.
If the issue is still relevant, please comment on the other issue.

Michele Moramarco
Prusa Research

closed 2023-01-12 by Prusa-Support