prusa3d/Prusa-Firmware-Buddy #2325: [BUG] Can't connect to SSID starting with space

Crocmagnon, 2022-07-30:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]

**Printer firmware version** - 4.4.0-BETA2
**Original or Custom firmware** - Original

**Optional upgrades** - Filament Runout Sensor

**USB drive or USB/Octoprint**
Not printing

**Describe the bug**
I can't connect the Prusa Mini to my wifi network. Its SSID starts with a space.

**How to reproduce**
* Create a wifi network with an SSID starting with a space
* Follow the [guide](https://help.prusa3d.com/guide/how-to-setup-wi-fi-mini-mini_316781) to setup the wifi on the Prusa MINI
* Configure your wifi credentials

I've tried the following files:
```ini
[wifi]
ssid= REDACTED SSID
key_mgmt=WPA
psk=REDACTED_PSK
```

```ini
[wifi]
ssid=" REDACTED SSID"
key_mgmt=WPA
psk="REDACTED_PSK"
```

**Expected behavior**
The printer should connect to the wifi network.

**G-code**
Not relevant.

**Crash dump file**
[dump.bin.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/9227182/dump.bin.zip)

**Additional comments**
I'm basing my assumption that it's linked to the initial space on the docs here: https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/215bcf5b377220655c831c730940324d4dbdbf47/lib/inih/ini.h#L42-L55

> Parse given INI-style file. May have [section]s, name=value pairs (whitespace stripped),

chillygithub, 2022-07-31:
Using spaces in the ESSID or PSK is _far from Best Practice_. And should be avoided, as its is known to break many implementations of the wireless stack.

Crocmagnon, 2022-07-31:
It may not be best practice but it’s definitely permitted by the standard 🙂

laloch, 2022-08-01:
Hi @Crocmagnon, thanks for reporting. While technically you're right that SSID can be any binary string including non-printable characters as long as it fits 32 octets, practically there are probably no consumer devices allowing to do so. We will certainly discuss the issue and my vote will certainly be for sticking with Cisco SSID policy which explicitly disallows leading and trailing whitespace characters.

Crocmagnon, 2022-08-01:
I guess using the BSSID would be fine 👍🏻

Crocmagnon, 2022-09-29:
Hi @laloch ! Thanks to the team for the RC1 release 😀 
I noticed that the release notes didn't mention the BSSID capability. Have you had time to discuss the feature?

Thanks 🙂

laloch, 2022-10-05:
Hi @Crocmagnon, we have the BSSID capability in our backlog. It is, however, scheduled for 4.5.0 at the moment.

Crocmagnon, 2022-10-05:
Thanks for the feedback! 👍🏻 Looking forward to testing this on my network.

Crocmagnon, 2023-07-24:
> it seems like the code would have to be trimming the field 

that’s what it does

nmschulte, 2023-07-24:
> that’s what it does

So I discovered; thanks.  I understand why this is the case now as well: https://en.wikipedia.org/wiki/INI_file#Keys_(properties)

lovemetwotimes, 2023-08-09:
SSID with spaces inside string also doesnt work.

nmschulte, 2023-08-17:
> SSID with spaces inside string also doesnt work.

SSIDs with internal spaces works fine for me.

douglsmith, 2023-12-19:
Thanks @nmschulte. I'm hoping your PR is implemented because I have a long-existing Wi-Fi network with a PSK that ends in a space. Changing the PSK is not really an option because it would require reconfiguring about 100 other devices and convincing others that the change is worth that pain. This would be just to accommodate one MK4 that can't deal with a character that is allowed in the Wi-Fi specifications. No other device on our network has ever had an issue with this in over a decade of use.

nmschulte, 2024-05-15:
> we have the BSSID capability in our backlog. It is, however, scheduled for 4.5.0 at the moment.

v4.5.0 has long since passed.  #3158 has received no attention.  @laloch, will this issue ever be resolved?

Crocmagnon, 2024-07-15:
> This issue has been flagged as stale because it has been open for 60 days with no activity. The issue will be closed in 7 days unless someone removes the "stale" label or adds a comment.

Oh please no not you Prusa 😔
I can come here every now and then to comment. I can even setup a script to automatically post a useless comment every now and then to keep this alive. No comment doesn’t mean the issue doesn’t exist anymore or people aren’t interested anymore. It just means that everything has been said and we’re waiting for a real implementation.

Either leave this open if you actually plan on implementing it someday or close it if you don’t but please, no stalebot 😩

Crocmagnon, 2024-07-15:
Sorry, I've been bit by organisations using stalebot to massively close issues in the past and I'm quite bitter about it.
I shouldn't have reacted like this.

Thank you for your answer ❤️