prusa3d/Prusa-Firmware-Buddy #2326: [BUG] Failing XYZ Axis test on v4.4.0 Beta

adynis, 2022-07-31:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - Mini
**Printer firmware version** - v4.4.0 Beta2
**Original or Custom firmware** - Original

**Describe the bug**
  With last firmware v4.3.4 XYZ test works fine. I change to v4.4.0, then XYZ test is failing. I reported this also to the support chat, and I re-did everything + 2 videos:
 - Re-flash v4.3.4
 - Reset default settings
 - Make a video with successful XYZ axis Test - https://www.youtube.com/watch?v=vE62S-qyzrc
 - Re-flash v4.4.0
 - Reset to default settings
 - Make a video with failing XYZ axis Test - https://www.youtube.com/watch?v=nOugUefCwT0

No other change was done between v4.3.4 and v4.4.0, but still the behavior of XYZ axis test is different. So I can not finalize the Wizzard, I can not "Auto Home" in v4.4.0

I write the topic also here since ... it looks like I'm the only one with this issue ? 
Is there anything else I could try to overpass this ?

murk-sy, 2022-07-31:
XYZ tests generate a `test_[serial]_xyz.txt` on the USB - if you provide both of those the devs can look into some more details.

adynis, 2022-07-31:
Unfortunately I get no file create on the USB on v4.4.0. I've attached only the v4.3.4 one, but ... Also after the Wizzard failing on the XYZ axis, ans also if I manually select Test XYZ axis, the result is the same: no file created on v4.4.0 on the USB :-( 

[firmware434 - test_CZPX2120X017XC08656_xyz.txt](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/9229680/firmware434.-.test_CZPX2120X017XC08656_xyz.txt)

I have no upgrade or any special change to the printer I can think on, which could influence the X or Y axis movement ...  Strange is that on v4.3.4 works immediately,..
Is anything else I could try to help debugging this issue ?

darraghbr, 2022-08-01:
The calibration wizard works fine on my machine upgraded to the 4.4.0 BETA. That is a very strange issue, might be worth a manual check of the belt tension but as far as I can see there was no changes to the selftest.

Prusa-Support, 2022-08-04:
Thank you for the report.

Reviewing the belt tension (possibly via tension meter), and cleaning and lubricating the smooth rods could actually help.
Please give it a try as per our instructions.
https://help.prusa3d.com/article/adjusting-belt-tension-mini_157324
https://help.prusa3d.com/guide/maintenance-tips_23200

Also as you (@adynis) pointed out, the mentioned issue (#2333) highlights that the motor's current rate and operation mode may change before or after a "move-axis" command. If this bug is confirmed on your printer too, you may want to take advantage of it: check how the axis test goes before a "move-axis" command (immediately after reboot) and after.

Michele Moramarco
Prusa Research

adynis, 2022-08-04:
Hi, Something IS there .... so after reading the experience from https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2333 I hardly succeeded to get a Pass for XYZ axis test, but after about 5..10 trials + using "move axis" before each of them.  And more than that, I succeeded to validate also the wizzard after another ~5 trials. But something is still fishy ... 
Right now I am able to have strange situations ... somehow it seems to me that the software considers immediately that the edges of the X or Y moving are touched after 1 mm , so the movement just stopps in both directions ...

I will try today or tomorrow more cases, maybe I can also do more videos which could help developers.

PS. I tried changing the tension of the belt, increase or decrease, but found no diff .... , I didn't yet try to lubricate the rods.
PS. Today, I even had a print on v4.4.0, after it started everything was fine, but ... before starting ... I had to again play ... "Move Axis", "auto home" (manually move the position), etc, until the auto mesh bed leveling before the print succeeded ...

Prusa-Support, 2022-08-10:
Apparently the "smooth movement mode" makes your printer more sensitive to friction on the axis, or something like that, but even after "opening move-axis" (to benefit from the bug, and change to "linear movement mode") you have mixed luck. So the linear movement helps but doesn't solve the problem.

According to issue #2333, the change of movement mode doesn't necessarily cause failures - I think it was only mentioned one case of incorrect auto-home.
While the different modes may affect the sensitivity of the axis movement, here there may seem to be a concern on the hardware side.

In the video [failing XYZ axis Test](https://www.youtube.com/watch?v=nOugUefCwT0), at minute 0:56', the "X-motor-visualizer" stops for a movement, after that the X-axis starts to move back. The movement doesn't seem interrupted though, so I don't know if was a problem with the visualizer only. Regardless of this, please keep on observing this happening on your future tests. 
You may need to further adjust the belt tension, and review the rightness of the motor pulleys.

Also, there seem to be accumulation of grease/dirt on the Y-smooth-rods. Maybe the smooth rods in general are not very clean. There could be marks and rough/sticky spots on the smooth rods.

In the case of a twisted or skewed axis, friction may increase too.

Michele Moramarco
Prusa Research

adynis, 2022-08-12:
Thanks for the details answer. I read it carefully and I did more test, but unfortunately with no positive result.
Here is another 5 minutes video with various tests: https://www.youtube.com/watch?v=OtibhUty63U
(PS. please ignore the orange rotation indicator from the X-axis motor; in this video the belts were looser, because I tried various variants, but I found no difference between my tests, so please ignore this aspect in this video)

Things I did:
 - removed X & Y belt, cleaned up X & Y axis, 
 - re-lubed a little bit the axis, 
 - checked that (without belt) rods are moving smooth on both axis (actually I checked that the bed (without belt) and the extruder (without belt) are moving smoothly
 - re-adjusted belt tension
 - set the Footer indicators on the LCD to be X-Axis, Y-Axis, Z-Axis
 - did various tests 
 - tighten the X/Y belts little bit more then normal + other tests
 - loosen the X/Y belts little bit more than normal + other test
 - did the above mentioned video

It's hard for me to find a pattern ... sometimes I get a correct "Auto-Home", but majority of times not, some times I can get one Axis-Test with green checkmark, but majority of time it;s a fail. So unfortunately I couldn't' formulate any helpful conclusion.

Any suggestion what else I could try ? (so that I can help you / the developers to identify the issue and come back with a solution for the ones like me )

darraghbr, 2022-08-12:
I'm not sure but this might be something that is being fixed by #2276

Prusa-Support, 2022-08-29:
Yeah, the pull request #2276 should make auto-home work more reliably, but we can't tell for sure if this will solve this specific issue. I guess we will see it in the next firmware release.

For now, the issue seems to be hardly reproducible. It seems the chances to see this issue are augmented by the smooth movement but, because the issue doesn't persist on all printers, it is apparently a consequence of a mechanical problem.
The mechanical problem could be caused by consumed bearings/rods or specifically sensitive motors/board.

I don't feel like recommending replacements based on tests that are run with a "non-stable" FW release.
Maybe, in a future release, acceleration, jerk and StallGuard values will be appropriately tuned to prevent these issues, were the hardware is not significantly fatigued.
Before even considering replacing any of these parts, I would recommend keeping on observing the phenomenon and monitoring similar cases posted on GitHub.

For now, homing manually (push by hand) should allow you to continue testing _FW 4.4.0-beta2_, or please downgrade the FW for hassle-free use of the printer until the next release.

Your feedback is most valuable. Further feedback will be appreciated.

Michele Moramarco
Prusa Research

adynis, 2022-08-29:
Hi, thanks for your answer. 
In the meantime I downgraded back to v4.3.4 because it was very hard to succeed to start a print (once it was started, it seemd to work). 
I was hoping for better news for me, but unfortunatelly it seems I have a more special issue :-( 
I will test imemdiately when a new firmware will appear;
If you have any suggestions what else could I test, **please let me know**, ( I'm willing to debug this because I was expecting the firmware with prussa connect since when Prusa Mini was launched :-)  )
For example if there are any test to see if the motors / bearings are ok in my case , or to check if things are moving smoothly or not .. (e.g.: can I check somehow the current consumed by the motors?) ... or any test which maybe could eliminate as much as possible my subjective interpretation or eventually my mistakes.

Prusa-Support, 2022-08-31:
Because the printer just works (with the previous firmware), I'm afraid the hardware will turn out to be within the specifics anyway.

With this beta firmware, something has clearly changed about the stepper motors operation, so there is a chance that this is only a matter of tuning some parameters.
Also, we are aware of issue #2333 and pull request #2276. A fix to the first, or the implementation of the latter, could help.

I'm sorry for the unfortunate situation. We will keep on monitoring similar cases and let the developers work on this.

Michele Moramarco
Prusa Research

adynis, 2022-09-30:
Heey.... IT WOORKS with v4.4.0-RC1 :thumbsup: :metal: ☀️ 
I've upgraded (from v4.3.4) to v4.4.0-RC1 which was released today. I tested it (without hopes, to be hones) and ... XYZ worked from the first trial without issues, like it was working on v4.3.4 !! I said ... maybe it's still .... based on 4.3.4 settings, or something. So I did a Reset to Defaults, then .... Wizzart ... and it worked without any glitch !

So... I'm not sure what exactly was the issue , if the developers removed any feature "because of me" ... but now I'm happy that I'm not anymore "the black sheep" (with not-working MINI on v4.4.0 :-p ) 

Thank you !

From my point of view this ticket can be marked as resolved.

Prusa-Support, 2022-09-30:
Thank you for your feedback.

I can speculate that this bug was not that widespread, but existed in very few specific hardware parts combinations - yet in compliance with the technical specifications.
Maybe this was not mentioned in the release notes because it only required minor polishing (not an actual bug fix).

Regardless of this, glad to hear that you are back in printing business and I'll be happy to close this case as solved.

Michele Moramarco
Prusa Research

closed 2022-09-30 by Prusa-Support