prusa3d/Prusa-Firmware-Buddy #2327: [FEATURE REQUEST] Binary-packing GCode when transferring over network

prochazkaml, 2022-07-31:
Please, before you create a new feature request, please make sure you searched in open and closed issues and couldn't find anything that matches.

If it makes sense, enter what type of printer or upgrade the feature request applies to.
**Printer type** - MINI, but could apply to other PrusaLink-compatible printers
**Optional upgrades** - not relevant

**Is your feature request related to an existing problem? Please describe.**
Transferring GCode over the network (via PrusaLink) is quite slow, especially on Wi-Fi, where it is apparent that the ESP interface is the largest bottleneck.

**Describe the expected functionality**
When the printer receives an API request for uploading files, it could accept an optional parameter which indicates that the input is binary-packed (perhaps according to the [RepRap draft spec](https://reprap.org/wiki/Packed_Binary_GCode) - quite nice, since it's streamable with little overhead on the depacker's side), with the benefit that the amount of data to be transferred is (sometimes significantly) smaller. Since most of 3D-printing GCode are X/Y/Z/E movements (→ decimal numbers), the above mentioned spec can pack them to BCD (2 digits → 1 byte), effectively reducing them in half.

Of course, PrusaLink would then have to implement an efficient packer that would run in the browser.

murk-sy, 2022-07-31:
Currently the limiting factor is not really the network, but more so the USB interface and all the other slog - though I'd like to see some form of compression added in the long run, mostly to reduce the file size and hopefully allow some checksumming to detect USB related issues.

laloch, 2022-08-01:
Hi @prochazkaml, thanks for your input. We are, indeed, already evaluating several ideas about binary encoding and specialized compression/decompression of the g-code text at each level of the processing stack (Slicer / browser / network / storage media) in order to improve both performance and stability.

prochazkaml, 2022-08-02:
Thank you so much for your replies! I'll respond to both of you in a single comment, since they both nicely complement each other:

@murk-sy:
> Currently the limiting factor is not really the network, but more so the USB interface and all the other slog

For the Ethernet interface, that might be true. However, if the Wi-Fi interface is 3-5x slower than Ethernet, then I'd argue that the Wi-Fi interface is still the most significant bottleneck:

> For now, the expected transfer speed from your computer to the printer should be around 250-300 KB/s over the Ethernet and 45-100 KB/s over the Wi-Fi.

(taken from the 4.4.0-BETA2 release notes)

Although, if the file _wasn't_ being unpacked on the printer side when receiving and it was stored in its packed form straight onto the USB stick, then we'll save USB bandwidth as well, making the transfer also faster for Ethernet, since the printer will be receiving and storing smaller files. A win for everyone, even for those not using PrusaLink, thanks to the smaller file sizes!

About the checksumming idea: while I agree with the concept, I'm not entirely sure that the printer can read a file of several MB from the USB interface in a reasonable amount of time before starting the print. Perhaps the packed GCode file could be split into smaller "chunks" (i.e. a few KB), each having their own checksum that would be checked during the print?

It's certainly a good idea to checksum a file that is entirely responsible for controlling your dangerously hot machine correctly, and a new binary format could definitely allow for that.

@laloch:
That's wonderful to hear! If PrusaSlicer did output the packed GCode directly, then there would be no need to do the packing in the web browser (I tried writing a simple routine in JS, and it took _a large amount of time_ to process a ~10 MB GCode file, so that might be a no go for most users), and the printer would then parse the binary packed GCode directly.

So, all in all, it would be (in my opinion) best if PrusaSlicer had an option for directly exporting packed GCode (based on the selected printer, perhaps) and the printer just parsed it directly, without the unnecessary conversion from binary to text GCode. But then there's the problem of supporting older firmwares, which do not accept anything except text GCode.

meermanr, 2022-10-04:
FWIW, I really like that G-code files are editable in a normal text editor.

I'd suggest using a compressed communications channel, and leave the files plain text. Similar to [HTTP Content-Encoding](https://developer.mozilla.org/en-US/docs/Web/HTTP/Headers/Content-Encoding) where ultimately the user receives the document verbatim, but the transport layer compresses at the sending side and decompresses at the receiver.

I expect that a standard compression algorithm such as [DEFLATE](https://en.wikipedia.org/wiki/Deflate) will work wonders on G-Code files - after all since G-Code is printable ASCII it already wastes the top bit of every byte. That's a 1/8th saving without doing anything clever.

A quick test on some G-code files I have lying around suggests that `gzip` will shrink the size anywhere between 2.5-9x.

And if you do decide to pack G-Code files, can you do something simple like `filename.gcode.zip` so ordinary users can easily decompress the original text for editing?

Prusa-Support, 2022-10-13:
Thank you all for your suggestions, our developers will take those into account.

Michele Moramarco
Prusa Reseaerch

closed 2024-06-19 by github-actions[bot]