prusa3d/Prusa-Firmware-Buddy #2328: [BUG] Fan warning chimes but does not appear if triggered when opening advanced menu

murk-sy, 2022-07-31:
**Printer type** - [MINI+]

**Printer firmware version** - [4.4.0-BETA2]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [Filament Runout Sensor]

**Describe the bug**
Fan warning chimes but does not appear if triggered when opening advanced menu, but only makes an audible chime. This may affect other warnings as well.

**How to reproduce**
1) Ensure fan check is enabled
2) Ensure #2314 is reproducible
3) Set print fan speed to 8/255 and wait for the fan warning message to start appearing. Different value under 40 may be required.
4) Go to settings menu
5) Press and hold the button until you hear/see another warning
6) Release the button and exit the advanced settings menu
7) Any further warnings do not appear, but you can hear the chimes
8) Printer must be restarted for warnings to appear again

**Expected behavior**
Warnings continue to be displayed normally or appear at least as a non-blocking notification

Prusa-Support, 2022-08-03:
Thanks for the report.

Are you talking about the [Experimental menu](https://help.prusa3d.com/article/experimental-menu-mini-mini_264396)?

Because, as you said, the issue may concern other warnings, please try to intentionally cause the triggering of a different error.
For example, carefully disconnect a thermistor from the Buddy board (avoid touching the board with bare hands or conductive objects) and a MINTEMP error should be displayed (unless the ambient is very warm).

Can you confirm this issue is not reproducible on _FW v4.3.4_ or older?

Michele Moramarco
Prusa Research

murk-sy, 2022-08-08:
@Prusa-Support Yes, this is when using the experimental menu.

I don't believe MINTEMP errors would affect this issue since those are BSODs that require printer restarts (I believe at least?)

The only similar warning/notice that I can think about would be the inactivity cooldown notification, which is difficult and time consuming to test with.
For the same reason it would be difficult to test on 4.3.4.

Prusa-Support, 2022-08-09:
Thank you again for your feedback.
The fan error bug should be fixed on the next release.

Michele Moramarco
Prusa Research

Prusa-Support, 2022-11-26:
The issue should be fixed by the final [FW 4.4.0](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.0) release now. Please check it out.

Michele Moramarco
Prusa Research

murk-sy, 2022-11-30:
The bug is still reproducible by setting speed to 8/255 and manually stopping the fan while reproducing step 5.

I don't believe it is a major issue that frankly anyone will encounter, but the bug should stay open and low priority. It is possible warnings being invisible will cause issues elsewhere with future features.

Prusa-Support, 2022-12-17:
Lowering the fan speed only to stop it by hand shouldn't be necessary.
Regardless of that, from the point of view of the official support, I suggest not touching the fan regardless of the speed.

Michele Moramarco
Prusa Research

murk-sy, 2022-12-17:
It's only to simulate a fan failure - fans on all my printers are all functioning flawlessly, so I just needed a reliable way to reproduce the issue. To clarify, the fan stop itself is not part of the issue, the on screen warning is.

Prusa-Support, 2023-01-03:
A 10% print fan speed won't bring benefit to your prints. Please consider disabling the fan below such value.

Stopping the fan by hand is not recommended and is not necessary for troubleshooting.
You may only want to reproduce the issue by moving the fan's cable around and see if power or RPM is affected: this helps tot reproduce an issue caused by a partially broken cable.
Obstacles and added friction of any kind on the fan's blades should be avoided: clean the fan in case of accidental errors and possibly don't touch it while operating.

Michele Moramarco
Prusa Research

murk-sy, 2023-01-07:
I'd like to reiterate the issue is NOT related to the fan or print settings at all - those are just steps to reproduce the issue by forcing the warning to appear. Disconnecting the fan instead of physically stopping it may also reproduce the issue.

The issue is specifically about the warning screen being permanently blocked from appearing (until restart) if triggered while opening the advanced menu.

Prusa-Support, 2023-01-20:
Oh, ok so I may seem to understand that the issue persists regardless of the speed of the fan.
I'll reopen the issue, thanks.

Michele Moramarco
Prusa Research

murk-sy, 2024-05-03:
Issue reproducible on 6.0.0, just need to select no when exiting experimental menu so the printer doesn't restart.

Opening "Move Z" menu doesn't appear to affect it, only the experimental settings menu does.
Additionally, it seems that doing this blocks future on-screen notifications as well as Connect notifications.

Still a very minor and unlikely issue, but I am not sure what other warnings/errors this would block as well. Keeping open for now.

murk-sy, 2024-08-27:
Haven't retested recently, but as said before it's a minor issue but it may block important warnings/notices from appearing.

murk-sy, 2024-11-02:
Reproducible on 6.2.0, updated instructions in first comment.
Once reproduced, this also blocks ATTENTION messages via Prusa Connect.

Cannot be reproduced with Z adjust (message pops up after closing the menu).

Not tested:
- If other messages can also cause the block
- If other long presses can cause the block
- If other menu openings can cause the block
- If other notifications (i.e. filament runout) do not work via Connect

CZDanol, 2025-01-06:
Thank you for your report. I've created an internal ticket BFW-6537 for this.