prusa3d/Prusa-Firmware-Buddy #2329: [BUG] After unloading filament, load must be selected twice

murk-sy, 2022-07-31:
**Printer type** - [MINI+]

**Printer firmware version** - [4.4.0-BETA2]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [Filament Runout Sensor]

**Describe the bug**
After unloading filament, the load option must be selected twice. Not a major issue but confusing for the end user.

**How to reproduce**
Unload filament and re-load it. This may be tangentially related to modified footer settings.
This also happens if you exit and re-enter the filament load menu.

**Expected behavior**
Go to load menu immediately

**Photo before load is pressed the first time**
[image]

**Photo before load is pressed the second time time (which then functions correctly)**
[image]

Prusa-Support, 2022-08-03:
Thank you for the report.

I wonder if the behavior persists on _FW v4.3.4_.

Would it make any difference if you restore the default footers, or perform a factory reset?

Does the autoload feature work normally instead? 
Does it make any difference whether the filament is fully removed and then inserted again (filament sensor goes "0" and "1" again), or unloaded and re-loaded without being removed (filament sensor stays "1" the whole time)?
Does it make any difference if the filament sensor is disabled or disconnected?

Michele Moramarco
Prusa Research

murk-sy, 2022-08-08:
@Prusa-Support 
Default footers do not seem to affect it, and this machine has already been factory reset.
If filament sensor is enabled, after pressing load once and then inserting the filament, the material selection screen pops up.
The behaviour of requiring 2 clicks appears to be the same if the filament sensor and autoload are disabled.

I practically always have filament loaded in the printer, so I also generally immediately load new filament after unloading a different one.

Prusa-Support, 2022-08-09:
Thank you for your feedback.
Our developers will look into this.

Michele Moramarco
Prusa Research

closed 2022-10-22 by murk-sy