prusa3d/Prusa-Firmware-Buddy #2330: [BUG] Cancelling remotely started print breaks remote printing start

murk-sy, 2022-07-31:
**Printer type** - [MINI+]

**Printer firmware version** - [4.4.0-BETA2]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [Filament Runout Sensor]

**Describe the bug**
After starting the print and cancelling it, exiting to the main menu causes the gcode file to be offered again - this also breaks further attempts at upload at print, returning a 409 conflict "Can't print right now"

**How to reproduce**
1) Upload and start print remotely
2) On the printer, cancel the print and select "Home"
3) The same print is offered to be printed again (as when the USB storage is plugged in or file was just uploaded)
4) Cancel this print as well
5) Try to upload and start print remotely (filename2)
6) Prusaslicer reports 409 Conflict "Can't print right now"; Filename2 is offered for printing
7) Automatic start printing does not work until the printer is restarted

**Expected behavior**
New print can be remotely started if the printer is not active

Prusa-Support, 2022-08-03:
Thank you for the report.

Please notice, as of now, per the [release notes](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.0-beta2), the printer <_must be on the home screen, “print-preview” screen, “print-finished” screen_> for Prusa Link to work correctly, and <_this is intentional for safety reasons_>.

I would say this is a known issue, or I should better say an intended behavior for _FW 4.4.0-beta2_.

Could you please confirm that the problem doesn't persist if you "stop print" via the PrusaLink web interface instead?

Michele Moramarco
Prusa Research

darraghbr, 2022-08-03:
duplicate of #2301

murk-sy, 2022-08-08:
@Prusa-Support This issue happens when the printer is on the home screen after cancelling. The steps taken are specified and shouldn't be skipped, please let me know if you cannot reproduce it. I have not tested web interface stop print as of yet.

Prusa-Support, 2022-08-16:
It seems we are looking at 3 different issues, which are not necessarily related.

1. Prusa Slicer (through Prusa Link) won't be able to 'upload & print' after stopping a print (unless the user presses the 'home' command). It seems the file will be transferred via Prusa Link, but Prusa Link won't be able to start the print.
2. When the user returns to the main screen ("home" command after stopping the print), the "latest" print file is offered - same as if the USB stick was just plugged.
3. When the USB stick is re-plugged (or point (2)) the last file automatically offered is not the latest uploaded via Prusa Link but the latest saved on the USB stick via computer.

Point (1) is actutally a known issue, because the printer <_must be on the home screen, “print-preview” screen, “print-finished” screen_> for Prusa Link to work correctly. The "print-stopped" case is probably treated differently than a “print-finished” one, even though I didn't have the chance to check if the problem persists in an actual “print-finished” scenario.

Point (2) is actually the case that the printer thinks that a flash drive is newly inserted, but it won't prevent the correct functionalityt of Prusa Link. I have reason to believe that the developers are aware of this behavior (I can't tell for sure), in fact the developers worked on allowing Prusa Link to do its job while the printer is on "print-preview" screen. Maybe this is not a bug but an enhancement request. 

I think the reason for the issue at point (3) to happen is that the files saved by Prusa Link on the USB flash drive are saved as "last modified" on a specific future date (check the picture below).
Maybe this deserves to be posted as a separate issue on [prusa3d/Prusa-Link-Web](https://github.com/prusa3d/Prusa-Link-Web) repository. I will only report to the developers for now

Michele Moramarco
Prusa Research
[image]

murk-sy, 2022-08-17:
@Prusa-Support 
>     1. Prusa Slicer (through Prusa Link) won't be able to 'upload & print' after stopping a print (unless the user presses the 'home' command). It seems the file will be transferred via Prusa Link, but Prusa Link won't be able to start the print.
> Point (1) is actually a known issue, because the printer <_must be on the home screen, “print-preview” screen, “print-finished” screen_> for Prusa Link to work correctly. The "print-stopped" case is probably treated differently than a “print-finished” one, even though I didn't have the chance to check if the problem persists in an actual “print-finished” scenario.

It is likely a different treatment, or cancelling just not resetting the printer to a "ready" state. It is possible that finishing one print normally after cancelling would restore upload+print functionality, but I haven't tested it.

>     2. When the user returns to the main screen ("home" command after stopping the print), the "latest" print file is offered - same as if the USB stick was just plugged.
> Point (2) is actually the case that the printer thinks that a flash drive is newly inserted, but it won't prevent the correct functionality of Prusa Link. I have reason to believe that the developers are aware of this behavior (I can't tell for sure), in fact the developers worked on allowing Prusa Link to do its job while the printer is on "print-preview" screen. Maybe this is not a bug but an enhancement request.

I've had this happen a few times already, but I'm not sure how to reproduce. I believe it's somewhat related to 3. as well.

>     3. When the USB stick is re-plugged (or point (2)) the last file automatically offered is not the latest uploaded via Prusa Link but the latest saved on the USB stick via computer.
> I think the reason for the issue at point (3) to happen is that the files saved by Prusa Link on the USB flash drive are saved as "last modified" on a specific future date (check the picture below). Maybe this deserves to be posted as a separate issue on [prusa3d/Prusa-Link-Web](https://github.com/prusa3d/Prusa-Link-Web) repository. I will only report to the developers for now

Imo, upload + present should not rely on latest file, but rather on the actual provided filename, just as upload+print does - would avoid a lot of issues.

Related issue is also #2331:
I currently have a corrupt USB drive that consistently exhibits that behaviour, but in a way so it always provides a specific, older file. However, in Windows file explorer, there is no visible issues (i.e. one file having a future date or similar), everything looks as it should.

I can prepare a raw data copy of that corrupt USB drive and privately provide access to it to the developers.

Prusa-Support, 2022-08-30:
My considerations were based on my own tests in the attempt to reproduce the issue, and the evidence I could collect so far, but feel free to run further tests and submit more feedback related to this specific issue.

In case of "upload+print" it doesn't matter how files are sorted, because the file being updated will be printed right away.
Nonetheless, as of now, it is important that the printer is on <_home screen, “print-preview” screen, “print-finished” screen_> for Prusa Link to correctly start the next print job, and an "interrupted print" is apparently not equal to a "finished print" scenario.
This behavior is intentional but may change in the future. Our developers are working on it.

We are analyzing the case explained issue #2331 too.
I recommend keeping the concern about the USB flash drive as a separate issue.

Michele Moramarco
Prusa Research

murk-sy, 2022-09-30:
4.4.0-RC1 appears to resolve automatic print start, closing.

closed 2022-09-30 by murk-sy