prusa3d/Prusa-Firmware-Buddy #2331: [BUG] Upload presents wrong file if USB storage is corrupt

murk-sy, 2022-07-31:
**Printer type** - [MINI+]

**Printer firmware version** - [4.4.0-BETA2]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [Filament Runout Sensor]

**Describe the bug**
In case of a USB file corruption of some sort, an older file can be presented for printing after upload. Upload and print functions correctly.
While this is mostly inconvenient, it can also lead to reprinting an older version of whatever you're trying to print - newer versions of the same file are likely to have long and similar names, which is common when developing a specific part.

**How to reproduce**
Difficult to pin down. Checking file modified/creation dates on Windows showed no obvious reasons for presenting a particular older file in the first place. Copying all files off, formatting the drive and copying them back resolved the issue.

**Expected behavior**
Remember the uploaded file name and present that file specifically for printing - do not rely on current standard "last added" file, which may be incorrect; or, use this to inform user of potential data corruption/date mismatches? on the USB storage.

The USB drive in printer 4 still has this particular issue - I am willing to provide a raw data copy (iso?) of the entire drive privately if requested.

Prusa-Support, 2022-08-04:
Thank you for your feedback.

In this case, the cornering aspect is the corrupted file or the faulty USB flash drive. A corrupted file or faulty USB flash drive is the anomaly itself and may be hard to reproduce.

Not sure if this is expected behavior but it truly depends on the compatibility of the corrupted file at this point.
We would like to look at the corrupted file itself if you are able to reproduce it, so that we can closely analyze it and observe the printer behavior with it.

If some allocation cells on the USB flash drive stopped working, there are programs and tools to restore the flash drive or discard corrupted cells - or a replacement USB flash drive would be all. 
In this case, there could be little to improve on the printer's FW side to help debug, and probably no way to restore and force the print of the files affected by the faulty USB flash drive.

Michele Moramarco
Prusa Research

Prusa-Support, 2022-08-17:
Not sure if we are looking at a corrupted flash drive or a corrupted file.
We may actually be looking at the collateral issue that emerged in issue #2330: wrong "creation/modificatiton" time if the file is uploaded over Prusa Link.
At this point, I would avoid mixing up the issues and close this one as a duplicate.

If we are actually looking at a different issue, which is a consequence of a faulty USB storage, it may not be of interest for our developers - unless it is proved that USB flash drives, in general, may fail in a very specific way, that can be reproduced and analyzed.
I would suggest closing the issue as "unable to reproduce", and keeping it closed as long as the failure is not reproducible.

I would close this issue if you don't mind, and it can be opened again in the future if there will be relevant evidence of a reproducible issue concerning (almost) all USB flash drives in general.

Michele Moramarco
Prusa Research

murk-sy, 2022-08-18:
The flash drive appears to function fine and Windows' default error checker does not find any issues. So maybe the drive is not even corrupt, but the printer doesn't like something about the files?

Here's a list of the top files by date. The highlighted one is presented when the USB drive is plugged in or if upload is triggered, despite there being plenty with newer dates.
[image]

**I have created a raw copy of the USB drive and was able to reproduce the issue exactly on a different USB drive and a different printer.**
I will provide the download to the USB drive raw image by email (it's 4GB compressed and contains company data I can't post publicly), along with other details on reproduction.

I will send an email to info@prusa3d.com and write your name and issue number in the email subject by the end of the day.

thmsbnrd, 2022-08-31:
I managed to fix this issue by removing all the gcode files from my USB and starting fresh. 
This might be because, even when sending a gcode via WiFi, the printer still presents the latest gcode file that you added manually to the USB. 
So by having only files sent via WiFi on the USB, it seems to fix this problem.

murk-sy, 2022-08-31:
I believe the issue was usage of a different sorting algorithm at home screen (for picking just the latest file) and that a fix has already been committed: https://github.com/prusa3d/Prusa-Firmware-Buddy/commit/4465b26410d89afe07f71da040a3d062b442b42a

The issue has actually appeared for me even before PrusaLink was available, but it made me take an extra look after seeing an incorrect file was presented after upload.

murk-sy, 2022-09-30:
4.4.0-RC1 appears to fix selecting the correct file on upload, but not when plugging in the USB drive

Prusa-Support, 2022-10-12:
Thanks for clarifying.

I'm glad that there was an improvement but, because the main issue or part of is still persistent, we could keep this issue open.

Michele Moramarco
Prusa Research

Prusa-Support, 2022-11-26:
The issue should be fixed in the final [FW 4.4.0](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.0) release now. Please check it out.

Michele Moramarco
Prusa Research

closed 2022-11-26 by Prusa-Support

murk-sy, 2022-11-30:
This one is difficult to reproduce intentionally, but will keep an eye out - it happened often enough that it shouldn't take long if it's still an issue.