prusa3d/Prusa-Firmware-Buddy #2332: [BUG] Assist sound mode does not work correctly in the Print menu

prochazkaml, 2022-07-31:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MINI+

**Printer firmware version** - 4.4.0-BETA2+4114

**Original or Custom firmware** - Original

**Optional upgrades** - none

**USB drive or USB/Octoprint** - USB flash drive

**Describe the bug**
When the Assist sound mode is active, it behaves oddly in the Print menu. When hitting the top/bottom of the list, it still produces the longer "error" beep, but moving the cursor around yields no sound effect being played. When I select a file to print, the sound effects start to work in the confirmation dialog, returning back to the file selection screen makes the sound effect disappear yet again.

**How to reproduce**
Turn on the Assist sound mode, enter the file selection screen and scroll around.

**Expected behavior**
The "click" sound should be heard everywhere when the Assist sound mode is active.

**Video**
https://prochazka.ml/prusa_issue_assist_sound_mode.mp4

Prusa-Support, 2022-08-04:
Thanks for the report.

Would the issue persist if you downgrade the firmware, or perform a factory reset?
https://help.prusa3d.com/guide/how-to-update-firmware-mini-mini_128421#128582
https://help.prusa3d.com/article/factory-reset-mini_177516

Michele Moramarco
Prusa Research

prochazkaml, 2022-08-04:
Hi. Performed a factory reset on 4.4.0, but the issue still remained.

Downgrading the firmware to 4.3.4 did resolve the issue, with the "click" sound now again present in the file selection menu.

Prusa-Support, 2022-08-09:
Hello! Thank you for this info, added the tag specific to this 4.4.0 Beta and relayed the info to devs, we'll look into it and fix it. 

Alessandro Pantaleo
Prusa Research

Prusa-Support, 2022-11-26:
The issue should be fixed by the final [FW 4.4.0](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.0) release now. Please check it out.

Michele Moramarco
Prusa Research

closed 2022-11-26 by Prusa-Support

prochazkaml, 2022-11-26:
Can confirm that this issue has already been resolved since the 4.4.0-RC1 release. Just flashed the final 4.4.0 release and everything still works as expected. Thank you! 👍