prusa3d/Prusa-Firmware-Buddy #2333: [BUG] Auto Home from the Calibration menu has inconsistent behavior

prochazkaml, 2022-07-31:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MINI+

**Printer firmware version** - 4.4.0-BETA2+4114

**Original or Custom firmware** - Original

**Optional upgrades** - none

**USB drive or USB/Octoprint** - USB flash drive

**Describe the bug**
When running the Auto Home upon booting up, it uses the same ease in/out movement that is now newly used in the Move Axis menu. Even after exiting the Calibration menu and entering it again, it does it like this. Only when I force axis movement via the Move Axis menu and then select Auto Home again, it now performs the regular linear movement on the axes. Oddly, this switch in behaviors also occurs when starting to print a GCode file and stopping it immediately before the file's auto home takes place.

One strange thing is that this manual auto home (once it switches to linear motion) appears to use more force on the motors than the auto home which is used when printing GCode.

I also encountered a strange issue when upon selecting Auto Home, it didn't actually home the X/Y axes at all (it just moved them slightly) and the nozzle went straight down into the middle of the plate. Of course, when I saw that it was going like that without even being properly homed, I immediately reset the printer. It only happened to me once and I couldn't get it to occur since.

**How to reproduce**
Power the printer up and select Auto Home from the Calibration menu. Then move the axes via the Move Axis menu and then try to perform auto home again.

**Expected behavior**
Auto home should use a single method for moving the axes.

**Video**
https://prochazka.ml/prusa_issue_auto_home.mp4

adynis, 2022-08-03:
Interesting, this might be related to "my issue" : https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2326
Some questions:
 1.) Have you successfully finished (all green) the calibration wizard ? 
 2.) Based on your video, what happens if you select Auto-Home multiple times, one auto-home after the other? (does the position maintain or change with each auto-home?)
 3.) Can you successfully Pass Calibration - XYZ-Axis Test ?

Prusa-Support, 2022-08-04:
Thank you for the report.

The difference in movement is hard to see for a such small move. Maybe starting in the auto home from about the middle of the XY axis would make the difference more visible in both cases.

I tried slowing down the video seed 0.25x to measure the time between the end of the heatbed movement towards the end, and the end of the heatbed movement towards the reference probing point.
It is impossible for me to measure time with accuracy, but I repeated a few times and there seem to be a slight difference.
I measured about 00:02.50'' for the "auto-home before move-axis" and 00:01.80'' for the "auto-home after move-axis".
An alternative quantitative comparison could be done by counting how many frames a specific movement takes, which I don't seem to be able to do.

Because the issue is difficult to see and objectively confirm, a "quantitative comparison" would be appreciated.

In addition to @adynis' questions (1) and (3), does the result change before and after the "move-axis" action is performed?
Does it make a difference if you singularly move the X or Y axis = would it fail X or Y axis but not both, if you move only on axis or the other?

Michele Moramarco
Prusa Research

prochazkaml, 2022-08-04:
Update: It turns out that it is not actually necessary to move any axes, this issue still triggers even if you just open the Move Axis menu and exit it immediately without changing anything. The issue affects both axes' movement, even if they are not moved at all.

@Prusa-Support:
I figured out that this issue also affects the Mesh Bed Leveling option in the Calibration menu, so I recorded that instead, as it is much more apparent: https://prochazka.ml/prusa_issue_mbl.mp4

(the Wizard prompt was caused by a factory reset, as was requested in issue #2332, but it makes no difference to the issue itself)

You can clearly see (and also hear, as each movement slowly ramps up/down before or aggressively accelerates after) the difference between both runs. I didn't record the Auto Home from a longer distance as requested, since this issue (at least I think) does not affect the speed, but rather the acceleration of the X/Y axes' movement, meaning that the time difference wouldn't be more noticeable than what I had posted before.

However, with Mesh Bed Leveling, the time difference is very noticeable, as it consists of many short movements, thus triggering the acceleration/deceleration more often:

Before entering the Move axis menu: **00:32.72"**
After entering the Move axis menu: **00:28.08"**

(started measuring when the printer was starting to tap onto the first point, stopped when it finished tapping on the last point)

I'd like to make something very clear here: this issue does not break any functionality of the printer (maybe except for auto-home, which failed on me once again today during testing, I'll try to make that particular issue reproducible). It's just odd that these routines (Auto Home, Mesh Bed Leveling and the XYZ test were tested, but others could also be affected) can behave differently based on a seemingly unrelated action (entering the Move Axis menu, printing a file or perhaps others).

@adynis:
1) Yes: https://prochazka.ml/prusa_issue_selftest.mp4 (I skipped the Z axis calibration after the selftest had finished, since I have done that already)
2) It seems as if the position is the same (both physically and in software - 176 X, 24 Y, 2 Z) between all runs.
3) Yes, see my response to question 1). I tried it separately (outside the calibration wizard) and it passed both before entering the Move Axis menu (with ease in/out motion) and after (with linear & quite aggressive movement).

Prusa-Support, 2022-08-09:
Thank you for your for getting back and for the clarification.

Our developers will look into this.

Michele Moramarco
Prusa Research

prochazkaml, 2023-01-29:
I can report that Auto Home no longer uses the slow acceleration (which I described as ease in/out earlier). However, after running Auto Home, moving the axes through the Move Axis menu and then running Auto Home again, I can hear a certain difference between the two runs (the second run of Auto Home has a bit more aggressive acceleration curve – when it starts to move an axis, it sounds as if it's trying to run the motors at constant speed without accelerating first, which causes the printer to shake a little bit) – although I'm not 100% sure.

I'll try to record some audio tomorrow, so I can visually tell by looking at the waveforms whether the issue really persists.

Prusa-Support, 2023-02-09:
A video where the noises could be heard would be appreciated.
Other than that - as long as the printer movements are not affected, and the difference in noise is not that noticeable - I wonder if the issue could be closed as solved.

Michele Moramarco
Prusa Research

prochazkaml, 2023-02-09:
Oh, sorry, I forgot to post, but in essence – my phone's microphone couldn't pick up anything (as it was de-noising, which I couldn't disable, I wanted to try another one, but I simply forgot about it).

The change in movement is very minor, and it does not affect regular printer functionality (as the auto home finishes successfully, and when printing, the auto home behaves normally), which means that this issue is so low-priority that it could be safely closed.

Prusa-Support, 2023-02-17:
We have evidence that something happened there in the last few firmware development stages, but we also have proof that the problem is fixed.

I'll close this issue now, feel free to share further feedback though.

Michele Moramarco
Prusa Research

closed 2023-02-17 by Prusa-Support