prusa3d/Prusa-Firmware-Buddy #2334: [BUG] Filament type check is not executed when printing remotely

Ferferite, 2022-07-31:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]

**Printer firmware version** - [4.4.0]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [none]

**USB drive or USB/Octoprint**
  usb/prusalink

**Describe the bug**
  When printing a file that was sliced for PETG for example and you have PLA loaded in your printer, a new warning pops up that states "this g-code was set up for another filament type". Very handy feature, but this check is not done when printing remotely.

**How to reproduce**
  Load any filament in the machine and slice the object with other filament settings, try to print from usb(get the warning), print from prusalink and the warning will not appear.

**Expected behavior**
  A warning appearing on the prusalink interface warning the object is sliced for another filament type.

**G-code**
 Not g-code specific.

**Crash dump file**
  -

**Video**
  -

Prusa-Support, 2022-08-04:
Thanks for the report.

This is more likely a feature yet to be implemented on remotely controlled prints.
Not sure if this is actually a bug, it sounds more like an enhancement.
I'd assign both labels and let the developers look into this.

Michele Moramarco
Prusa Research

Prusa-Support, 2022-08-09:
The Filament type warning via Prusa Link is expected to be implemented in one of the next few updates.

Michele Moramarco
Prusa Research

NesporBellisRM, 2023-01-17:
I would classify this as a bug ( not an enhancement ) - this is not expected behaviour and is different from if you start exactly the same file from the physical control panel. The print will start even if there is no filament loaded if you submit and start a print via PrusaLink.

Expected behaviour - if there is no filament loaded then the print should not start.

Prusa-Support, 2023-01-24:
I don't mind adding a "bug" label but please notice this is yet to be implemented, and it is already planned to be implemented.

Nothing "broke" and this is not necessarily an unintended behavior. This is something yet to be implemented at this early development stage. It will be implemented during the "natural course" of the Prusa Link development. 
Thanks for understanding.

Michele Moramarco
Prusa Research

closed 2024-06-19 by github-actions[bot]