prusa3d/Prusa-Firmware-Buddy #2340: 25 point mesh bed level

seifer69er, 2022-08-02:
I have found an issue with the x axis rods that corresponds to a bad first layer.

The right hand side of the print bed is being over compensated due to the inflection point of the x axis rods due to the nature of the gantry.

The only way to overcompensate this is to either increase the angle of the x axis wrt the bed plate, or increase the mesh bed probe points to a 5x5 array that is closer to the rhs.

There is a large offset between measurements 1 8 9 and 16 wrt the rhs. The lhs has the correct amount of spacing to the edge of the bed.

I have been speaking with Marek from prusa about this, however a 5x5 first layer bed mesh level would fix this issue.

Otherwise its more hardware tweaks as per above.

Eldenroot, 2022-11-26:
This could solve a lot of issues with bad assembling of machine... I mean the problem with Z axis (wombling).

Prusa-Support, 2022-12-26:
Thanks for sharing your idea.

The MK3 mesh bed leveling proved to be already quite reliable with the default 3x3 probing points grid.
For the more demanding scenarios, the MK3 mesh bed leveling was improved with the optional 7x7 grid, for a total of 49 probing points, 11 of which should be discarded and simply interpolated due to the closeness of magnets.

The 4x4 grid on the MINI's quite small bed is a good tradeoff: less dense than 7x7 (optional MK3 grid) but much denser than 3x3 (optional MK3 grid). A denser mesh should also consider the interference caused by magnets in such a reduced area.

The current mesh should be sufficient and will work best on a well-straightened printer geometry. 
Improving the printer's geometry after the final assembly remains our best advice.

A denser mesh on the MINI could be a bit tricky to be implemented and yet it may not guarantee the best possible first layer (not without minor geometry straightening).
It definitely sounds good though.

The issue will stay open so that our developers can make their considerations on it.

Michele Moramarco
Prusa Research

Prusa-Support, 2022-12-26:
I beg you pardon. 🙇
This issue is actually a duplicate of https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/1264, and therefore will be closed. 
Feel free to share your thoughts on the already existing issue.

Michele Moramarco
Prusa Research

closed 2022-12-26 by Prusa-Support