prusa3d/Prusa-Firmware-Buddy #2347: [BUG] Cannot load filament after using "Change filament", screen flickers/flashes

SpeakingOfBrad, 2022-08-02:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - Mini+

**Printer firmware version** - 4.4.0-BETA2

**Original or Custom firmware** - Original

**Optional upgrades** - Filament Runout Sensor

**USB drive or USB/Octoprint**
  USB/PrusaLink

**Describe the bug**
  Trying to use Change Filament to change to a new filament, the filament unloads properly but the screen flickers/flashes during the "Was filament unload successful?" screen. I can click yes, but then I get prompted to "Please remove filament from filament sensor." even though there is no filament in the sensor. There is no way to get past this screen except for rebooting the printer.

**How to reproduce**
  In the Filament menu, select Change filament.

**Expected behavior**
  Unload filament, then get prompted to load new filament.

**G-code**
  N/A

**Crash dump file**
  N/A?

**Video**
  https://user-images.githubusercontent.com/824333/182419323-d09f9081-6b56-4838-ad73-2d8d3798b2b7.mp4

laloch, 2022-08-02:
Hi @SpeakingOfBrad, thanks for reporting. I can (sort of) confirm the issue. For me the display starts flashing, but I can still control the printer and finish the filament change. Once I'm done, the display stops flashing and everything's back to normal.

SpeakingOfBrad, 2022-08-02:
Thanks for the update! It turns out my filament sensor isn't working... But I don't think that has anything to do with this bug.

MistrKernnunos, 2022-08-03:
Try to disable the filament sensor. It should let you load the filament

darraghbr, 2022-08-03:
So is this a duplicate of #2299 ?

SpeakingOfBrad, 2022-08-03:
> So is this a duplicate of #2299 ?

Yep, sorry. I didn't see that one.

closed 2022-08-03 by SpeakingOfBrad