prusa3d/Prusa-Firmware-Buddy #2354: [BUG] <Live Adjust Z value frozen>

Matej1023, 2022-08-04:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI+]

**Printer firmware version** - [e.g. 4.4.0, ...]

**Original or Custom firmware** - [Original]

**Optional upgrades** - []

**USB drive or USB/Octoprint** - USB flash drive 

**Describe the bug**
  After some time, the Live Adjust Z value freezes. However, it is clear that the sensor is working correctly because it lights up and the printer performs Mesh Bed Leveling correctly. But it is interesting that the Live Adjust Z value is displayed correctly in the Prusa Link interface.

**How to reproduce**
  Hard to say. I have no idea why this happened. The value froze in the middle of the print at -1.6 mm.

**Expected behavior**
  It does not affect the function of the printer.

**G-code**
  Any one G-Code

Prusa-Support, 2022-08-04:
Hello! Thank you for the report. How did you notice more specifically it was frozen mid-print? Did you refer to the Live-Z adjust set during first layer calibration, or more the current Z position of the printer? Feel free to send me a video of this to better review, we'll look into it. Thank you! 

Alessandro Pantaleo
Prusa Research

Matej1023, 2022-08-04:
> Hello! Thank you for the report. How did you notice more specifically it was frozen mid-print? Did you refer to the Live-Z adjust set during first layer calibration, or more the current Z position [...]

The value in the bottom line that shows the current height of the Z axis has frozen. Unfortunately, I cannot send you the video, as I needed to print large objects that take around twenty hours to print, I needed to be sure that the printer will print the entire object in order, so I have now played the firmware to version 4.3.4. After printing I can flash the firmware back to 4.4.0 and if the problem occurs again I can send a video. I'm still sending a photo of the display showing what value is frozen so we can talk about the same thing. 
[image]

closed 2024-06-19 by github-actions[bot]