prusa3d/Prusa-Firmware-Buddy #2358: [BUG] Screen Flashing on Filament change process

kghunt, 2022-08-04:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]

**Printer firmware version** - [e.g. 4.4.0 Beta.]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [WiFi]

**USB drive or USB/Octoprint**
  USB or Wifi

**Describe the bug**
  When selecting filament change the screens flashes during the whole process but returns to normal once the filament change is complete.

**How to reproduce**
  Do a filament change using the menu.

**Expected behavior**
  Screen does not flash on an off.

**G-code**
  n/a

**Crash

https://user-images.githubusercontent.com/5150662/182875337-00f42ec5-6181-46f5-a7ca-22c024d69ebc.mp4

 dump file**
  n/a

**Video**
  attached

columna1, 2022-08-04:
Can confirm, I have the same issue on my printer.

prochazkaml, 2022-08-05:
As @JohnnyDeer said over on #2299:
> This bug will be fixed in next release.

AJErazzor, 2022-08-08:
Confirm same issue - not quite every time

Prusa-Support, 2022-08-09:
Hello everyone! Thank you for the report, I confirm the reply from @JohnnyDeer on the other thread, it will be fixed on the next release - closing this particular thread as a duplicate of https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2299. 

Alessandro Pantaleo
Prusa Research

closed 2022-08-09 by Prusa-Support