prusa3d/Prusa-Firmware-Buddy #2359: [FEATURE REQUEST] Allow file rewrite for PrusaLink uploads

jkavalik, 2022-08-04:
When I try to reupload a file that already exists on the flashdrive, I currently get an error (403 iirc?). PrusaLink on MK3+RPi does not do that and rewrites the file instead. I find that very useful because it allows me to restart prints (after first-layer problems etc.) by just re-"upload and print" directly from PrusaSlicer instead of opening the PrusaLink in a browser and searching for the right file to start.

So would it be please possible to implement that on the Mini too?
- automatically rewrite already existing file if a file with the same name is uploaded.

chrishilbert, 2022-08-09:
The filename's unique portion is the print time by default.  I often make minor settings changes that do not alter the print time.  As a result, I cannot upload the new file without first going into the UI and deleting it first.  Annoyance.

Prusa-Support, 2022-08-17:
Hello! Thank you for the report and sorry for the late response from my side. I'll move the issue to the Prusa Link repository for now, so that it will be on the attention of our Prusa Link devs, in case of any more info from our side we'll let you know here, and, in case of any more feedback on this aspect, anyone feel free to further comment on this thread. 

Alessandro Pantaleo
Prusa Research