prusa3d/Prusa-Firmware-Buddy #2364: [ENHANCEMENT] Make API key entry a password field and/or set an authentication cookie

RichiH, 2022-08-07:
Prusa Mini, Filament sensor & ESP01S installed, v4.4.0-beta2

Connecting to Prusa Link Beta requires an API key through a clear text field. Browser do not autodetect this as a password field, so they do not offer to save the API key. Neither does Prusa Link Beta set a cookie with the API key.
This means that I have to look up and enter the API key every single time.

For more convenience, please
1. make the API key entry a password field, and/or
2. set an authentication cookie.

Bazimek, 2022-08-08:
Yes please!

chrishilbert, 2022-08-10:
Agreed to both suggestions.  Would be nice if it was linked to Prusa Slicer some how.  Like I could launch Prusa Link FROM Prusa Slicer with the configured IP/API key sent over automatically.  That wouldn't require a cookie or anything if you could just submit it to the form.