prusa3d/Prusa-Firmware-Buddy #2366: [ENHANCEMENT] Park head higher on Z axis when doing filament change

reedy, 2022-08-07:
Please, before you create a new feature request, please make sure you searched in open and closed issues and couldn't find anything that matches.

Enter what type of printer or upgrade the enhancement applies to.
**Printer type** - MINI
**Optional upgrades** - Filament Runout Sensor

**Is your suggested improvement related to an existing problem? Please describe.**
Not sure

**Describe the expected functionality**
Don't know whether this is a request that belongs here, or over at https://github.com/prusa3d/PrusaSlicer (though I haven't noticed a setting), but when you do a filament change mid print, the head is parked in the bottom left (nearest the front), but not at a very high Z axis height. When you're doing a big colour change (for example, Black to White), it's useful to be able to clean some of the residue off the nozzle. Unfortunately, with the default height, it's pretty difficult to do that as there's not much room to get even a small brush like [these spark plug bruses](https://ffx.co.uk/product/Get/Lessman-Les125111-4017359003312-Spark-Plug-Brush-2-Row-Brass-Wire)

Will get a photo when the next change comes up in a bit (though that's white to black, but still representative).

reedy, 2022-08-08:
[image][image][image]

Prusa-Support, 2022-08-13:
Thank you for the suggestion. It sounds like a legit request.
I'll make sure to bring this to the attention of our developers.

Michele Moramarco
Prusa Research

Prusa-Support, 2022-09-01:
This behavior was already improved by PR https://github.com/prusa3d/Prusa-Firmware-Buddy/pull/1063, solving issue https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/637.

This issue is actually a duplicate but requests a further enhancement on top of the previous one, so I'd leave it open for now.

Michele Moramarco
Prusa Research

closed 2024-06-19 by github-actions[bot]