prusa3d/Prusa-Firmware-Buddy #2367: [BUG] It it no longer possible to assign a static IP address to the printer in 4.4.0-BETA2

kellertk, 2022-08-07:
**Printer type** - Prusa Original MINI+

**Printer firmware version** - 4.4.0-BETA2

**Original or Custom firmware** - Original

**Optional upgrades** - Filament sensor

**USB drive or USB/Octoprint**
Both

**Describe the bug**
The Prusa Connect Local (PrusaLink) docs talk about the [Save/Load Settings option](https://help.prusa3d.com/article/prusa-connect-local-mini_133279#save-load-settings) in order to assign a static IP to the printer, because it's not possible to do so from the display. However, in the new firmware this menu item is missing - it's been replaced with the Wi-Fi Save/Load credentials option. This new option doesn't actually set IP addresses and won't work for the Ethernet interface anyway.

**How to reproduce**
1. Update to 4.4.0-BETA2.
2. Try and find the Save settings item option in the Ethernet settings menu.
3. Not there.

**Expected behavior**
I don't have a DHCP server running on my printer network. I can set one up, but that's a real pain. I'd much rather assign addresses to the printer and the other devices on that network statically, just because it eliminates DHCP being down as a reason that my printers might drop off network. Please either restore the Save/Load Settings option in the Ethernet settings, or find out how to do IP entry with the rotary encoder :)

I have to go back to 4.3.4 for now.

**G-code**
N/A

**Crash dump file**
N/A

Prusa-Support, 2022-08-13:
Thanks for the heads up.
It seems the feature was accidentally removed from the UI. This will be reported to our developers.

Michele Moramarco
Prusa Research

kurzschlussidi, 2022-11-28:
Hey 
i just tried it on the released 4.4.0 Version and the option is still not showing.

vorelus12, 2022-12-08:
Thank you for reporting, I have same problem with set static IP.

kellertk, 2022-12-25:
Any updates on this issue? This is a significant regression since 4.3.4, and because of this missing feature I can't upgrade my printer's firmware.

Prusa-Support, 2023-01-05:
I'm sorry, sadly no news.
Our developers will look into this at their earliest convenience.

Michele Moramarco
Prusa Research

kurzschlussidi, 2023-01-19:
Why are #2838 and this thread now closed?

jvlobo, 2023-03-30:
Any updates on fixing the static IP address configuration?
My router doesn't allow to set static IP address to my devices and since I can't also not set it on the printer itself, every time I turn on the printer, a new IP address is set, so I have to go to the network menu to check the IP address that was assigned, and then go to PrusaSlicer and modify the printer's address, quite a big pain after having to do it so many times 😭

maciejbaur, 2023-03-30:
In 4.4.1 I have the same issue and the only workaround I have is to set static DHCP address for my MINI+ in ruter DHCP server.

jvlobo, 2023-03-30:
If you are able to set that in your router is the only option at the moment. Sadly I can't do it as my router is heavily restricted by the internet company. My only option would be to buy my own router.

Prusa-Support, 2023-04-21:
The router settings can surely be an option.

I'm sorry about this.
We don't have an estimated time for the fix but I'd expect this to be solved sometime soon, as we plan to enable Prusa MINI on Prusa Connect.

Michele Moramarco
Prusa Research

jvlobo, 2023-04-26:
🙏🏽 Thanks.
Hopefully it can be fixed soon as set it on the router is not an option for me sadly :(

Prusa-Support, 2023-11-01:
I can't confirm if there was a bug in the previous firmware version but there is a new firmware version (alpha) to be tested and assigning a static IP address is possible.

_1) Write down the current printer’s IP address, mask, and gateway
2) Reserve the IP address in your router settings! (We can not help with this step, depends on router model or IT has to be contacted in bigger networks)
3) Switch the IP to static in menu network -> Wifi or Ethernet or both..
4) If not done before, in Wifi menu select “create credentials”
5) Plug the USB drive to the computer and edit the prusa_printer_settings.ini (as a text file)
6) Fill in the IP and other info. You can fill in both Ethernet and WiFi at the same time as well
7) Plug the USB back to printer and if using Wifi, go to “load credentials” in wifi menu
8) Go to settings -> system -> load settings on MK4/XL or settings -> load settings from file on MINI to load this info to the printer_

```
[eth::ipv4]
type=static
addr=YourReservedIpAddress
mask=YourMask
gw=YourGateway
[wifi]
ssid=YourWifiName
psk=SuperSecretWifiPassword
[wifi::ipv4]
type=static
addr=YourReservedIpAddress
mask=YourMask
gw=YourGateway
```

**Again, as per step [2], we can't help with setting up the router but Prusa Connect is an alternative solution to access your printer (or find its IP).
Thanks to the last firmware upgrade (alpha version, may be unstable) Prusa MINI is newly enabled on Prusa Connect too!**

Can this issue be closed?

Michele Moramarco
Prusa Research

jvlobo, 2023-11-01:
Why can't the static IP be set as in any other WiFi enabled device? I can set a static IP address on my smartphone or any computer I have without going into the router.
Sadly I can't reserve or assign static IP to devices on my router because it is controlled by the company. My only option would be to either use Prusa Connect or buy another router. But I don't understand why such a standard functionality that is available in basically every WiFi device is that difficult to get on the printer (I'm probably missing something but nobody explains why this is such a difficult task so I can only make assumptions....)

Thanks!

LukasRajter, 2023-11-16:
Dear Prusa support,

Setting static IPs works. Could I ask you to verify whether the same procedure remains effective when changing static IPs multiple times? For instance, after setting certain static IPs, you then decide to set different ones.

The first time, I successfully set static IPs, but my next attempt to use different static IPs was not successful (I completely repeated your procedure). I still see the same static IPs as the first time.

Firmware version: 5.1.0-alpha2+13126

Thank you
Lukas

LukasRajter, 2023-11-24:
Hey,

I have still the same issue, as was mentioned in my previous comment, with the 5.1.0 stable release. Even, if I did full factory reset - no success with changing of static IPs.

The following works to me:
- downgrading firmware from 5.1.0 to 5.1.0-alpha1, set there static IPs for wifi and eth. I followed instructions mentioned by Prusa support in this thread so content of prusa_printer_settings.ini file is:

 ```
    [eth::ipv4]
    type=static
    addr=YourReservedIpAddress
    mask=YourMask
    gw=YourGateway
    [wifi]
    ssid=YourWifiName
    psk=YourWifiPassword
    [wifi::ipv4]
    type=static
    addr=YourReservedIpAddress
    mask=YourMask
    gw=YourGateway
```

- upgrading to 5.1.0-alpha2 and then finally back to 5.1.0

Prusa-Support, 2023-11-29:
@jvlobo I don't have experience with the automatic static IP assignment as this has never been a fully automatic procedure for our printers as far as I know. If you believe this process can be automated, please consider opening a separate issue about that as a feature request.

As per the step 
> 2) Reserve the IP address in your router settings! (We can not help with this step, depends on router model or IT has to be contacted in bigger networks)

we don't expect you to buy another router but contact your company's IT team for help with this matter.

.

@LukasRajter If you are trying to set a custom IP, I'm not sure if that is possible.
The steps listed above require you to take note of the _IP address, mask, and gateway_ that were automatically assigned to the printer, so I'm not sure if you can deliberately modify them but you should be able to make them "permanent".
Maybe I'm wrong here, maybe I didn't understand the problem correctly and I'm not an IT expert myself, but probably the router's customer support can help you set a custom IP.

Michele Moramarco
Prusa Research

jvlobo, 2023-11-29:
> @jvlobo I don't have experience with the automatic static IP assignment as this has never been a fully automatic procedure for our printers as far as I know. If you believe this process can be [...]

> Michele Moramarco Prusa Research

Sadly I can't set custom static IP addresses on my router. That's the whole point... you can set a static IP in almost any device that connects to your local network. You can do that in any computer, smartphone, tablet... etc etc, from the device itself just going into network settings, selecting static IP and writing your 192.168.1.whatever. 
I wish this was possible on the Prusa Mini.

MightyPork, 2023-12-11:
I believe I used static IP on the mini before, in 4.4, but it forgot the settings after upgrading. None of the guides with the INI file worked.

I know for sure the esp8266 supports static IP, I used that in several projects myself. It's just a bug in the Buddy firmware.

Currently I solved it by setting a static DHCP lease on my PiHole which I use as a DHCP server.

Prusa-Support, 2024-05-04:
A new Help article is out.
https://help.prusa3d.com/article/how-to-set-a-static-ip-address_695648

Our knowledge base can't cover the topic thoroughly yet as this feature partially relies on the router settings, and our support clearly limited there. However, advanced users can set a static IP and consider reaching the router manufacturer' support for additional help.

Other than this, Prusa Connect is a valid alternative to actively monitor and fully control your printer - or find the current IP.

I'd like to take this chance to suggest closing this issue, which, besides the considerations above, has been silent for months now.

Michele Moramarco
Prusa Research

MightyPork, 2024-05-04:
Is the bug fixed, or you just added a note about setting DHCP address reservation on a router?

That isn't usable for people with cable routers like the one provided by T-Mobile, they don't let you configure anything.

I have relied until now on the fact that the router seems to assign the address for a very long time, but it's not too trustworthy and the address is also different from what I used before this firmware upgrade, when static address setting worked.

jvlobo, 2024-05-04:
I think they didn't really understand, or maybe whoever replied last from Prusa-Support didn't read the whole issue (I don't blame him/her) but in the documentation they link, it actually explain how to set static IP address for both WiFi and Ethernet, basically you need to add this to your `prusa_printer_settings.ini` file (inside your USB drive):

```
[wifi::ipv4]
type=static 
addr=192.168.0.42 
mask=255.255.255.0 
gw=192.168.0.1 

[network] 
dns4=1.1.1.1;8.8.8.8
```

And then as they explain in the documentation:

> Copy this file to a USB drive and upload it to the printer via the menu Settings > System > Load Settings from File. This will load all the settings from the .ini file. (Note, this is a different menu than "load settings" for Wi-Fi settings, which only uploads Wi-Fi data.) Afterward, restart the printer.

I just did this and I was finally able to have the static IP address working 👏🏽

Thanks!

jvlobo, 2024-05-04:
I also tried whatever was in that part of the documentation in the past and it didn't work for me either.
I'm not sure what they updated, it may be this last part:

> For setting a static IP address for Wi-Fi instead of the Ethernet, the process is exactly the same, but instead of [eth::ipv4] use [wifi::ipv4].

I don't remember the `[wifi::ipv4]` part and my printer is connected via WiFi (like most people probably).

MightyPork, 2024-05-11:
Today I tried this on the stable 6.0.0. release and the above example with `wifi::ipv4` and static IP does not work at all, it just says "Link down" and stays like that.

Switching to DHCP makes it work immediately with a dynamic address.

jvlobo, 2024-05-11:
I did not touch anything on the printer itself, I just put the file on the SD card, loaded as they explain on the documentation and it works. I still keep the same IP address I put on the file.
My version is `5.1.0` actually. Now I'm scared to update 🤔

MightyPork, 2024-05-11:
> I did not touch anything on the printer itself, I just put the file on the SD card, loaded as they explain on the documentation and it works. I still keep the same IP address I put on the file. My version is `5.1.0` actually. Now I'm scared to update 🤔

Did you make a backup copy of the file? I'd like to make my test with a known to work file - but so far my router consistently gives the printer the same IP, so I can live with DHCP.

Here is my exact file that was accepted but didn't work (creds redacted):

[prusa_printer_settings.txt](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/15282515/prusa_printer_settings.txt)

jvlobo, 2024-05-12:
This is mine: [prusa_printer_settings.txt](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/15286422/prusa_printer_settings.txt)

I think it looks basically like yours, so I guess the only difference is the firmware version on mine, very strange.

danopernis, 2024-06-27:
Let me close this as aduplocate of https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3549

closed 2024-06-27 by danopernis