prusa3d/Prusa-Firmware-Buddy #2371: [BUG] Previewing gcode with ¤ (┬ñ) in Filename completely stops the printer

murk-sy, 2022-08-08:
**Printer type** - [MINI+]

**Printer firmware version** - [4.4.0-BETA2]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [Filament Runout Sensor]

**Describe the bug**
If a file is sent via Prusalink with the character "¤" (or perhaps others), it completely locks up the printer, even Prusalink access. A reset is necessary.

[image]

**How to reproduce**
1) Start a print via Prusalink with filename `test¤.gcode`.
2) The file is saved on the USB drive as `test┬ñ.gcode`.
3) The printer is completely locked up and inaccessible, even via Prusalink
File can be browsed, but as soon as the preview is opened, the printer locks up
Note that the file itself is a perfectly valid gcode file - only the filename has non standard characters.

**Expected behavior**
Avoid problematic filenames completely or do not accept non standard characters.

kellertk, 2022-08-08:
Related: #2368

Prusa-Support, 2022-08-17:
Thank you for reporting.

The bug is not related to Prusa Link, in fact it persists even if a file including an incompatible character is directly transferred to the USB flash drive, directly via computer.

In the past, some special characters would cause a "blue screen crash" (BSOD) and this was resolved for some specific characters.
Probably - as you said - there are more incompatible characters, and probably any incompatible character in the file name would freeze the printer during the print preview as of the current firmware version.

I will bring this to the attention of the developers.

Michele Moramarco
Prusa Research

murk-sy, 2022-09-30:
RC1 improves on this behaviour:
Upload+Print appears to start the print normally.
Navigating via USB allows you to select a print, however it closes the preview window immediately without displaying the image preview. Additionally, plugging in the USB drive appears to ignore the file when selecting the latest one.
I'd call that good enough for now, but will keep issue open for devs to decide.

Prusa-Support, 2022-10-05:
Thank you for your feedback, and the issue will stay open.

Michele Moramarco
Prusa Research

murk-sy, 2024-05-03:
Can not reproduce anymore on Mini FW 6.0.0 with filename `test┬ñ.bgcode`, closing

closed 2024-05-03 by murk-sy