prusa3d/Prusa-Firmware-Buddy #2375: [BUG] Crash bug when printing over Prusa Link Beta

RichiH, 2022-08-08:
**Printer type** - MINI

**Printer firmware version** - 4.4.0-BETA2+4114

**Original or Custom firmware** - Original

**Optional upgrades** - Filament Runout Sensor, ESP01S. **NB: While I have the ESP01S installed, I printed over Ethernet**

**USB drive or USB/Octoprint** - Sent via Prusa Slicer 2.5.0-alpha3 to Prusa Link; USB thumb drive installed in Mini for storage
  
**Describe the bug**
I printed successfully two times, with the third attempt I got this error:
[image]

when trying to print https://www.printables.com/model/254416-print-in-place-tolerance-tester

**G-code**
[Print in Place Tolerance Test Sliders_0.2mm_PETG_MINI_1h30m.gcode.txt](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/9282661/Print.in.Place.Tolerance.Test.Sliders_0.2mm_PETG_MINI_1h30m.gcode.txt)

RichiH, 2022-08-08:
I restarted my Prusa Mini and tried again. Same gcode, same error. I have switched it off for safety, for now.

[image]

RichiH, 2022-08-08:
I ran OCR over it so Google can pick it up and so others looking for similar issues can find it:

TASK: tcpip_thread. Illegal
unaligned load or store
bot: 0x2001b0e0 top: 0x2001ca14
CPUID:410fc241 ICSR :0043d803
VTOR :00020200 AIRCR: fa050300
CCR  :00000210 HFSR :40000000
DER :09199000 CPACR:00f00000
0x00000001 0x00000000 0x00000000
0x00000000 0x00000000 0x00000000
0x00000000 0x00000000 0x00000000
0x00000000 0x00000000 0x00000000
0x00000000 0x00000000 0x00000000
0x00000000 0xffffffed 0x00000003
0x000003e8 0x00000001 0x00000001
0x000003e8 0x20013d20 0x00000000
0x00000001 0x00000001 0x080237db
0x2001acd0 0x000081c1 0x080cef94
0x00000000 0x000081c1 0x2000f5dc
0x000081c1 0x00000000 0x00000001

4.0-BETA2+4114

kellertk, 2022-08-08:
Same as #2342?

RichiH, 2022-08-08:
The first one yes, I didn't find that, sorry... I can enable the crash dump
and try again until I reproduce.

Thanks!

Sent by mobile; please excuse my brevity.

chrishilbert, 2022-08-10:
Looks like a dupe of #2315

RichiH, 2022-08-11:
Thanks @chrishilbert & @kellertk!

In the second image in this issue, there's a different error message. As such, I won't close this issue as a duplicate of either #2315 or #2342 but devs should feel invited to do so in case they determine them to be the same.

closed 2024-06-19 by github-actions[bot]