prusa3d/Prusa-Firmware-Buddy #2376: [FEATURE REQUEST] [MINI] A way to pause a print and resume at a later time (with other prints in between)

andyolivares, 2022-08-09:
I have a MINI+ and I've been using it with great success. One feature that would be nice is the possibility to "permanently" pause a print and then resume it at a later time.

I know, it is possible to pause/resume, but in the same print, but that's not exactly what I'm suggesting.

Due to the fact that some prints take really long times, and that bed sheets are interchangeable, it would be nice to have the option to "permanently" pause a print, change the bed sheet, do some other prints, and then get back to the longer print (still in the other bed).

I've faced some cases where I'm running a very long print and suddenly would like to finish a smaller print, but can't do that and have to wait for the longer print to finish.

Having this feature would be a great addition to this great printer.

murk-sy, 2022-08-09:
Generally, by the time you would finish some other print, the bed sheet would cool down and the print would release from it. And if not that, you will probably bend the sheet when removing it, causing the print to pop off or at least partially detach.

Assuming the print stays stuck *and* you've got a spare sheet ready to home onto, you could actually do this now. However, you would have to do some manual gcode editing. Alternatively, you can split prints ahead of time or recover partially failed (or stopped early) prints by doing the same.

Adding this feature and maintaining it anywhere near a reliable level is not worth the effort in my opinion.

Prusa-Support, 2022-08-26:
Hello! Thank you for the thread, as @murk-sy mentioned, with the feature there, the issue would be in keeping the removed sheet heated so that the print won't pop off - likely this feature won't be implemented like this, although in a completely different way we imagined this "multitasking" in our Prusa Pro AFS, shown at the Dubai Expo https://afs.prusa3d.com/. 

Alessandro Pantaleo
Prusa Research

closed 2024-06-19 by github-actions[bot]