prusa3d/Prusa-Firmware-Buddy #2377: Allow alternative graphic to octoprint octopus

AJErazzor, 2022-08-09:
Would be a great enhancement to "pimp" the mini with a different graphic to the octoprint octopus by selecting your own

kellertk, 2022-08-14:
I would argue that it's not really correct to include the octoprint octopus in the firmware anyway, since you can serial print with software that isn't octoprint.

Prusa-Support, 2022-08-18:
Thank you for the suggestion.

We got similar requests, such as requests for showing additional information while printing via Octoprint (#189) and for displaying a generic "serial connection" instead of the Octoprint logo if no-Octorpint connected (#2008).

I wouldn't close this issue as a duplicate because this on is more specifically a request for UI customization.
Maybe modding the firmware on your own would be the only way to accomplish that, but I'll take notes for now, and let our developers review.

On a side note, the currently missing Octoprint image is a known issue and apparently intentional (#2300). This will be fixed soon.

Michele Moramarco
Prusa Research

AJErazzor, 2024-05-23:
Could this show information from the internet/prusalink or some other configurable source

mattolenik, 2024-08-19:
I'd also like to see this