prusa3d/Prusa-Firmware-Buddy #2387: [QUESTION] Starting SD print over Gcode

pauhull, 2022-08-12:
Hi,

I am trying to start a SD print with Gcode commands, specifically M23, M26 and M24, as per Marlin and Prusa documentation. After sending the commands the printer won't start a print or even respond with an error. This is how my console looks like in Pronterface:
```
Connecting...
Printer is now online.
>>> M115
SENDING:M115
FIRMWARE_NAME:Prusa-Firmware-Buddy 4.3.4 (Github) SOURCE_CODE_URL:https://github.com/prusa3d/Prusa-Firmware-Buddy PROTOCOL_VERSION:1.0 MACHINE_TYPE:Prusa-mini EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Cap:SERIAL_XON_XOFF:0
Cap:BINARY_FILE_TRANSFER:0
Cap:EEPROM:0
Cap:VOLUMETRIC:1
Cap:AUTOREPORT_TEMP:1
Cap:PROGRESS:0
Cap:PRINT_JOB:1
Cap:AUTOLEVEL:1
Cap:Z_PROBE:1
Cap:LEVELING_DATA:1
Cap:BUILD_PERCENT:0
Cap:SOFTWARE_POWER:0
Cap:TOGGLE_LIGHTS:0
Cap:CASE_LIGHT_BRIGHTNESS:0
Cap:EMERGENCY_PARSER:0
Cap:PROMPT_SUPPORT:1
Cap:AUTOREPORT_SD_STATUS:0
Cap:THERMAL_PROTECTION:1
Cap:MOTION_MODES:0
Cap:CHAMBER_TEMPERATURE:0
>>> M20 L
SENDING:M20 L
Begin file list
XYZCAL~1.GCO
End file list
>>> M23 XYZCAL~1.GCO
SENDING:M23 XYZCAL~1.GCO
>>> M26 S0
SENDING:M26 S0
>>> M24
SENDING:M24
>>> M27
SENDING:M27
Not SD printing
```

As you can see M27 is reporting that there is no print running. Is there anything I am missing? Is SD card printing over Gcode simply not supported by default (this is the stock firmware from the website)? Also, why is M20 only showing the DOS 8.3 filenames although the L parameter is supplied?

Thank you in advance :)

Prusa-Support, 2022-08-27:
Hello! Thank you for your message and sorry for the late response. I took a first look at the documentation and noticed that these commands would require "SD Support" specifically, which I am unsure the program Pronterface has (it's designed to print via USB) - if I understood correctly, you are printing via USB but taking the GCode from an SD card inserted into the PC? I believe the issues are stemming from this "compatibility" between "SD Support" and Pronterface itself, both for the M20 command but also the other commands with the read file and start print itself. I'll try investigating it more and in case of more info I'll update here. 

Would it work regularly with Pronterface but taking the GCode from PC itself? 

In case of anything to further review feel free to comment of course. 

Alessandro Pantaleo
Prusa Research

pauhull, 2022-08-27:
Hello,

actually I was just using Pronterface for demonstration purposes. I am sending the commands over serial via a script, so Pronterface has nothing to do with the problem here.

I'm sorry if I created confusion by calling it SD printing. I am trying to print a file that is located on the USB stick that is plugged 
directly into the printer. I just say SD printing because that's what the Marlin documentation calls it.

What do you mean with "SD Support"? I looked through documentations from Marlin, Prusa and on the RepRap wiki but couldn't find any further information. I only know that the printer reports its capabilities (the lines starting with "Cap:") to the host when queried with a M115 command. Is there a way for a host to report capabilities on Prusa firmware?

And yes, regular printing over serial works fine. I just write the gcode commands to the serial port and wait for the "ok" response to send more.

arekm, 2023-04-19:
M20 worked (partially, no "L" parameter handling in MK4 firmware, a regression compared to MK3) with usb medium.

That means that cardreader interface (lib/Marlin/Marlin/src/sd/cardreader.cpp) in MK4 firmware partially works with USB.

M20 calls
````
card.ls();
````

M23 calls
````
card.openFile(parser.string_arg, true);
````

from the same cardreader interface. Some developer needs to take a look why ls() works there while openFile() fails.

Prusa-Support, 2024-04-16:
Code `M23` should work now as per recently merged PR https://github.com/prusa3d/Prusa-Firmware-Buddy/pull/3766.
The same pull request also adds support for code `M32`, which theoretically allows selecting and starting a print file in one go.

Please give it a try with the final firmware version [6.0.0](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.0.0) which was just released.

Michele Moramarco
Prusa Research

Prusa-Support, 2024-06-28:
It seems safe to close this issue as solved.

Michele Moramarco
Prusa Research

closed 2024-06-28 by Prusa-Support