prusa3d/Prusa-Firmware-Buddy #2389: [FEATURE REQUEST] <Enter comprehensive title>

HK-Moebius, 2022-08-12:
It would be more convenient if the api-key could be saved in a text file on the USB stick. This way you would not need to type the key by hand. A QR code on the display would also be a nice solution.

chrishilbert, 2022-08-13:
There's already a request out there to cookie it.

Prusa-Support, 2022-08-30:
Hello! Thank you for your message and sorry for the late response from my side. I'll link this thread to the one on Prusa Link where there was the request to cookie it https://github.com/prusa3d/Prusa-Link-Web/issues/222, closing this specific thread as related with that one, but in case of anything more feel free to add more comments of course. 

Alessandro Pantaleo
Prusa Research

closed 2022-08-30 by Prusa-Support