prusa3d/Prusa-Firmware-Buddy #2392: [BUG] M105 no longer reporting correct "A" temperatures on 4.4.0-BETA2

kellertk, 2022-08-13:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - Prusa Original MINI+

**Printer firmware version** - 4.4.0-BETA2

**Original or Custom firmware** - Original

**Optional upgrades** - Filament sensor

**USB drive or USB/Octoprint**
Both

**Describe the bug**
In 4.3.4, `M105` reported Hotend, Bed, Ambient, Hotend power, and Bed Power.
```
Send: M105
Recv: ok T:32.87 /0.00 B:29.80 /0.00 A:43.00 /0.00 @:0 B@:0
```

In 4.4.0-BETA2, `M105` returns bogus Ambient values.
```
Send: M105
Recv: ok T:32.65 /0.00 B:30.00 /0.00 A:-30.00 /0.00 @:0 B@:0
```

Prusa-Support, 2022-08-26:
Hello! Thank you for the report, I passed this on to our devs for them to review and fix it asap. In case of more info from our side we'll post on this thread. 

Alessandro Pantaleo
Prusa Research

pauhull, 2022-11-26:
This is still the case in 4.4.0+4180.
[image]

Prusa-Support, 2022-12-09:
Hello, thank you for letting us know, reported to devs. Hopefully it can be fixed asap on a future release. 

Alessandro Pantaleo
Prusa Research

closed 2024-06-19 by github-actions[bot]