prusa3d/Prusa-Firmware-Buddy #2393: [BUG] <Enter comprehensive title>

Hauzman, 2022-08-13:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]

**Printer firmware version** - [4.4.0 beta 2]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [ Filament Runout Sensor]

**USB drive or USB/Octoprint** -[ USB ]
  Please let us know if you print via USB flash drive or USB/Octoprint

**Describe the bug**
  A clear and concise description of what the bug is.
In the main menu it apeare the nozzle heating 2 time,
The second one is in the right down corner i this this do not need to be there 

**How to reproduce**
  Please describe steps to reproduce the behavior.

**Expected behavior**
  A clear and concise description of what you expected to happen.

**G-code**
  Please attach the G-code you had trouble with. This will make it easier for us to replicate the error.

**Crash dump file**
  Please attach the crash dump file. This will make it easier for us to investigate the bug.

**Video**
  Please attach a video. It usually helps to solve the problem.

kellertk, 2022-08-13:
In the beta firmware, the footer is customizable. You need to go into the [experimental menu](https://help.prusa3d.com/article/experimental-menu-mini-mini_264396) to change it.

Hauzman, 2022-08-14:
Hello Sorry I did not know i got the mini a few day
Thanks for the information 👍

closed 2022-08-14 by Hauzman