prusa3d/Prusa-Firmware-Buddy #2394: [BUG] translations missing in DE, ES language

hertzi, 2022-08-14:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MINI

**Printer firmware version** - v4.4.0-beta2

**Original or Custom firmware** - Original

**Optional upgrades** - Filament Runout Sensor

**USB drive or USB/Octoprint**
  USB flash drive + Wifi

**Describe the bug**
  For Prisa Link a couple of translation keys are missing in some non English translations (at least DE, ES) (prop.nozzle-diameter, proj.add-from.title, proj.add-from.local, upld.start-pt, printer.title, upld.direct.choose, proj.details, proj.download

**How to reproduce**
  Switch Prusa Link in version 4.4.0 Beta2 to DE or ES language

**Expected behavior**
  All fields, buttons and labels should be translated if the language of Prusa Link is switched from EN to DE.

Prusa-Support, 2022-08-26:
Hello! Thank you for the report, I brought it to our devs attention so that it can be reviewed and fixed as soon as possible. 

Alessandro Pantaleo
Prusa Research

Prusa-Support, 2022-10-14:
Hello.
Are these errors actually appearing on the Prusa Link web UI?
If so, I would consider moving this issue to [prusa3d/Prusa-Link-Web](https://github.com/prusa3d/Prusa-Link-Web) repository.

Michele Moramarco
Prusa Research

hertzi, 2022-10-15:
@Prusa-Support you are right, this report is about Prusa Link web UI. Sorry for the confusion.

michalrudolf, 2024-07-10:
Issue is not relevant to this repository.

closed 2024-07-10 by michalrudolf