prusa3d/Prusa-Firmware-Buddy #2401: PID Autotune enabled in GUI

FrankCellini, 2022-08-16:
the printer can do PID auto tune however it would be good to add this function to user interface so that it can be easily done and saved in “configuration.h” file.

Enter what type of printer or upgrade the enhancement applies to.
**MINI**
**PID Autotune**
**Unable to set different PID values when altering hotted configuration eg adding silicon sock, changing heat break**
**Easily do PID Autotune and set new values in “configuration.h” file**

rbronco21, 2022-09-27:
I read that PID tuning work work in custom g-code. Is that true? I'd be happy running an autotune (from menu or g-code) that only output PID values and then manually entering them into my g-code.

FrankCellini, 2022-09-27:
Yes PID Auto tune is a function you can do on printer and this gives you values that you can then put into G Codes on a print job in Slicer.  You have to connect to printer serial port and use terminal software on a connected computer.
[image]

closed 2024-06-19 by github-actions[bot]